micropython/micropython #5686: ESP23 - Scanning Bluetooth Characteristics results in Kernel Panics

JohannesNeu, 2020-02-22:
Hello,

I have modified some bluetooth code to read the characteristics of a xiamoi flower meter (hardcoded a mac address for my first tests). I can scan the devices and list the services using the interrupt method described in the documentation. But as soon as I try to retrieve the characteristics I get a kernel panic.

This is the code I use:
`import time
import bluetooth
import binascii
from micropython import const

_IRQ_CENTRAL_CONNECT                 = const(1 << 0)
_IRQ_CENTRAL_DISCONNECT              = const(1 << 1)
_IRQ_GATTS_WRITE                     = const(1 << 2)
_IRQ_GATTS_READ_REQUEST              = const(1 << 3)
_IRQ_SCAN_RESULT                     = const(1 << 4)
_IRQ_SCAN_COMPLETE                   = const(1 << 5)
_IRQ_PERIPHERAL_CONNECT              = const(1 << 6)
_IRQ_PERIPHERAL_DISCONNECT           = const(1 << 7)
_IRQ_GATTC_SERVICE_RESULT            = const(1 << 8)
_IRQ_GATTC_CHARACTERISTIC_RESULT     = const(1 << 9)
_IRQ_GATTC_DESCRIPTOR_RESULT         = const(1 << 10)
_IRQ_GATTC_READ_RESULT               = const(1 << 11)
_IRQ_GATTC_WRITE_STATUS              = const(1 << 12)
_IRQ_GATTC_NOTIFY                    = const(1 << 13)
_IRQ_GATTC_INDICATE                  = const(1 << 14)

def bt_irq(event, data):
    if event == _IRQ_CENTRAL_CONNECT:
        conn_handle, addr_type, addr = data

    elif event == _IRQ_CENTRAL_DISCONNECT:
        conn_handle, addr_type, addr = data

    elif event == _IRQ_GATTS_WRITE:
        conn_handle, attr_handle = data

    elif event == _IRQ_GATTS_READ_REQUEST:
        conn_handle, attr_handle = data

    elif event == _IRQ_SCAN_RESULT:
        addr_type, addr, connectable, rssi, adv_data = data
        print(addr)
        if addr == b'\xc4|\x8df\xdd\xf0':
            print(data)
            bt.gap_connect(addr_type, addr, 2000)

    elif event == _IRQ_SCAN_COMPLETE:
        pass

    elif event == _IRQ_PERIPHERAL_CONNECT:
        conn_handle, addr_type, addr = data
        bt.gattc_discover_services(conn_handle)

    elif event == _IRQ_PERIPHERAL_DISCONNECT:
        conn_handle, addr_type, addr = data

    elif event == _IRQ_GATTC_SERVICE_RESULT:
        conn_handle, start_handle, end_handle, uuid = data
#        print("conn_handle: " + str(conn_handle))
#        print("start_handle: " + str(start_handle))
#        print("end_handle: " + str(end_handle))
#        print("uuid: " + str(uuid))
        if str(uuid) == "UUID16(0xfef5)":
            bt.gattc_discover_characteristics(conn_handle, start_handle, end_handle)

    elif event == _IRQ_GATTC_CHARACTERISTIC_RESULT:
        conn_handle, def_handle, value_handle, properties, uuid = data
        print(data)

    elif event == _IRQ_GATTC_DESCRIPTOR_RESULT:
        conn_handle, dsc_handle, uuid = data

    elif event == _IRQ_GATTC_READ_RESULT:
        conn_handle, value_handle, char_data = data

    elif event == _IRQ_GATTC_WRITE_STATUS:
        conn_handle, value_handle, status = data

    elif event == _IRQ_GATTC_NOTIFY:
        conn_handle, value_handle, notify_data = data

    elif event == _IRQ_GATTC_INDICATE:
        conn_handle, value_handle, notify_data = data

bt = bluetooth.BLE()
bt.active(True)
bt.irq(handler=bt_irq)
bt.gap_scan(10000, 10000, 10000)

time.sleep(20)
`
and this is the error message i see in repl:
`Entering REPL. Use Control-X to exit.
>
MicroPython v1.12 on 2019-12-20; ESP32 module with ESP32
Type "help()" for more information.
>>> 
>>> import irqtry
I (232800) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
b'(\xff<\xb5#.'
b'\xc4|\x8df\xdd\xf0'
(0, b'\xc4|\x8df\xdd\xf0', True, -64, b'\x02\x01\x06\x03\x02\x95\xfe\x15\x16\x95\xfeq \x98\x00\x00\xf0\xddf\x8d|\xc4\r\x07\x10\x03\x00\x00\x00')
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40151b34  PS      : 0x00060430  A0      : 0x80158bdb  A1      : 0x3ffbb530  
A2      : 0x3ffc6e2c  A3      : 0x00000000  A4      : 0x3f4286fe  A5      : 0x3ffc2b94  
A6      : 0x00000003  A7      : 0x3ffc2b68  A8      : 0x00000000  A9      : 0x3ffbb4f0  
A10     : 0x3ffc6e2c  A11     : 0x00000000  A12     : 0x00000004  A13     : 0x3ffbb530  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40151b31:0x3ffbb530 0x40158bd8:0x3ffbb560 0x40153af9:0x3ffbb5a0 0x40155f31:0x3ffbb5d0 0x40153781:0x3ffbb600 0x40153793:0x3ffbb620 0x401526a6:0x3ffbb640 0x400f78c7:0x3ffbb660 0x40097615:0x3ffbb680

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4988
load:0x40078000,len:10404
load:0x40080400,len:5680
entry 0x400806bc
I (519) cpu_start: Pro cpu up.
I (519) cpu_start: Application information:
I (519) cpu_start: Compile time:     Dec 20 2019 07:56:38
I (522) cpu_start: ELF file SHA256:  0000000000000000...
I (528) cpu_start: ESP-IDF:          v4.0-beta1
I (533) cpu_start: Starting app cpu, entry point is 0x40083014
I (526) cpu_start: App cpu up.
I (544) heap_init: Initializing. RAM available for dynamic allocation:
I (551) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (557) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (563) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (569) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (575) heap_init: At 3FFCC8A0 len 00013760 (77 KiB): DRAM
I (581) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (588) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (594) heap_init: At 40099FB8 len 00006048 (24 KiB): IRAM
I (600) cpu_start: Pro cpu start user code
I (619) spi_flash: detected chip: generic
I (619) spi_flash: flash io: dio
I (619) cpu_start: Chip Revision: 1
W (621) cpu_start: Chip revision is higher than the one configured in menuconfig. Suggest to upgrade it.
I (632) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
MicroPython v1.12 on 2019-12-20; ESP32 module with ESP32
Type "help()" for more information.
>>> 
`

My tests were done with esp32-idf4-20191220-v1.12.bin on a JZK® ESP-32S ESP32 Development Board as well as a TTGO ESP32.

codefreax, 2020-02-22:
I was able to repro with my build. It is a NULL-ptr deref deep in nimble:

```
> xtensa-esp32-elf-addr2line -aipfC -e ~/Documents/Coding/esp/micropython/micropython-src/ports/esp32/build-GENERIC_SPIRAM/application.elf 0x40159115:0x3ffd96c0 0x4016187e:0x3ffd96f0 0x40161321:0x3ffd9730 0x4015bad5:0x3ffd9760 0x4015a395:0x3ffd9790 0x4015a3a7:0x3ffd97b0 0x4015fd12:0x3ffd97d0 0x400f91a7:0x3ffd97f0 0x400948bd:0x3ffd9810
0x40159115: ble_gattc_rx_read_type_adata at ??:?
0x4016187e: ble_att_clt_rx_read_type at ??:?
0x40161321: ble_att_rx at ble_att.c:?
0x4015bad5: ble_hs_hci_evt_acl_process at ??:?
0x4015a395: ble_hs_process_rx_data_queue at ??:?
0x4015a3a7: ble_hs_event_rx_data at ble_hs.c:?
0x4015fd12: nimble_port_run at ??:?
0x400f91a7: ble_host_task at nimble.c:?
0x400948bd: vPortTaskWrapper at port.c:?
```

```
(gdb) disassemble 0x40159118
Dump of assembler code for function ble_gattc_rx_read_type_adata:
   0x40159100 <+0>:	entry	a1, 48
   0x40159103 <+3>:	l32r	a11, 0x4014ed80
   0x40159106 <+6>:	extui	a10, a2, 0, 16
   0x40159109 <+9>:	mov.n	a13, a1
   0x4015910b <+11>:	movi.n	a12, 4
   0x4015910d <+13>:	call8	0x401589e0 <ble_gattc_extract_with_rx_entry>
   0x40159110 <+16>:	mov.n	a2, a10
   0x40159112 <+18>:	beqz.n	a10, 0x40159124 <ble_gattc_rx_read_type_adata+36>
   0x40159114 <+20>:	l32i.n	a8, a1, 0
   0x40159116 <+22>:	mov.n	a11, a3
   0x40159118 <+24>:	l32i.n	a8, a8, 4
   0x4015911a <+26>:	callx8	a8
```

My interpretation of the disasm is that `rx_entry` is NULL at `rc = rx_entry->cb(proc, adata)`:
```
  void
  ble_gattc_rx_read_type_adata(uint16_t conn_handle,
                               struct ble_att_read_type_adata *adata)
  {
  #if !NIMBLE_BLE_ATT_CLT_READ_TYPE
      return;
  #endif

      const struct ble_gattc_rx_adata_entry *rx_entry;
      struct ble_gattc_proc *proc;
      int rc;

      proc = BLE_GATTC_RX_EXTRACT_RX_ENTRY(conn_handle,
                                           ble_gattc_rx_read_type_elem_entries,
                                           &rx_entry);
      if (proc != NULL) {
          rc = rx_entry->cb(proc, adata);
          ble_gattc_process_status(proc, rc);
      }
  }
```

Reading an attribute works, if I do every step manually in the REPL. I can repro the crash only if I call functions in the IRQ handler like in your example. Not sure if ubluetooth is meant to be used that way..? @jimmo, @dpgeorge can you comment on that?

dpgeorge, 2020-02-24:
> Not sure if ubluetooth is meant to be used that way..?

Yes you should be able to call BLE functions in the IRQ handler itself.  But it could be that it's not working properly in some cases.

jimmo, 2020-02-24:
Thanks for the report @JohannesNeu and for digging into it in such detail @codefreax .

As @dpgeorge says, this usage is definitely supported.

I wasn't able to repro the exact failure, but I was able to get a very similar crash which is fixed by #5690.

@JohannesNeu, this unrelated, but instead of writing:

```
if str(uuid) == "UUID16(0xfef5)":
```

you should be able to use

```
_THE_UUID = bluetooth.UUID(0xfef5)
...
   if uuid == _THE_UUID:
```

(relying on the str representation of things is a bit fragile).

JohannesNeu, 2020-02-24:
Thanks for your help! How can I test the changes?

jimmo, 2020-02-24:
@JohannesNeu here's a firmware.bin built from this PR https://drive.google.com/open?id=1UfrIWo957yxAji7c2sOuhmWEI9qZ_Iig  (it's for the generic non-spiram build, which should be equivalent to what you're using currently)

JohannesNeu, 2020-02-24:
@jimmo I tested your firmware with the code I mentioned in the initial post and I sadly get another kernel panic:
```
>>> import test
I (193096) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 2
b'(\xff<\xb5#.'
b'\x1bP4.\xdc\xd2'
b'(\xff<\xb5#.'
b'\x1bP4.\xdc\xd2'
b'\xc4|\x8df\xdd\xf0'
(0, b'\xc4|\x8df\xdd\xf0', True, -58, b'\x02\x01\x06\x03\x02\x95\xfe\x15\x16\x95\xfeq \x98\x00\xc4\xf0\xddf\x8d|\xc4\r\x07\x10\x03\x00\x00\x00')
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40155880  PS      : 0x00060230  A0      : 0x8015c3ed  A1      : 0x3ffbb4c0  
A2      : 0x3ffc6760  A3      : 0x00000000  A4      : 0x3f4244bd  A5      : 0x00000003  
A6      : 0x3ffc29d4  A7      : 0x3ffc29e0  A8      : 0x00000000  A9      : 0x3ffbb480  
A10     : 0x3ffc6760  A11     : 0x00000000  A12     : 0x00000004  A13     : 0x3ffbb4c0  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x4015587d:0x3ffbb4c0 0x4015c3ea:0x3ffbb4f0 0x40159261:0x3ffbb530 0x4015b296:0x3ffbb560 0x40152d75:0x3ffbb590 0x40152d87:0x3ffbb5b0 0x40156436:0x3ffbb5d0 0x400f7bef:0x3ffbb5f0 0x40092d29:0x3ffbb610

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5176
load:0x40078000,len:12856
load:0x40080400,len:3480
entry 0x40080638
I (525) cpu_start: Pro cpu up.
I (525) cpu_start: Application information:
I (525) cpu_start: Compile time:     Feb 24 2020 17:33:23
I (528) cpu_start: ELF file SHA256:  0000000000000000...
I (534) cpu_start: ESP-IDF:          v4.0
I (539) cpu_start: Starting app cpu, entry point is 0x400832d0
I (531) cpu_start: App cpu up.
I (550) heap_init: Initializing. RAM available for dynamic allocation:
I (557) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (563) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (569) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (575) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (581) heap_init: At 3FFCC7E8 len 00013818 (78 KiB): DRAM
I (587) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (594) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (600) heap_init: At 4009EA78 len 00001588 (5 KiB): IRAM
I (606) cpu_start: Pro cpu start user code
I (625) spi_flash: detected chip: generic
I (625) spi_flash: flash io: dio
I (625) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
MicroPython v1.12-196-ge9eeb2470 on 2020-02-24; ESP32 module with ESP32
Type "help()" for more information.
>>> 
```

AussieSusan, 2020-03-01:
I get the same situation. I'm using esp32_idf4-20200227-v1.12-195-gb16990425.bin from the Micropython repository running on a esp32-wrover-b.
The code I'm using is basically the ble_temperature_central.py sample program with the exception thatI modified the device it is looking for to one that I've written.
(BTW - the reason I have gone this way is that I was getting almost exactly the same thing - a kernel panic - when using ESP-IDF V4, the 'bluetooth' stack and what was basically the same code as the Espressif 'gattc_demo.c' software. It makes me wonder if there is not an issue with the Espressif BLE code somewhere deep down in the stack that is common to both of their BLE stacks.)
Susan

tve, 2020-03-01:
If you could produce a source backtrace, that might be helpful....
- build the firmware using `DEBUG=1` option to make
- grab https://github.com/tve/esp32-backtrace 
- when you get that guru meditation error, you run the backtrace script with `build-XXX/application.elf` as arg and paste the error and backtrace line in (the git repo has an example). 

That will tell you/others what the call stack was at the point of crashing...

AussieSusan, 2020-03-01:
Happy to do that but my lack of knowledge of how to carry out those instructions is showing...
When you say 'build the firmware' as this is a python script, how do I add that option?
I have copied the script but as I am using Atom as the editor, how to I create the xxx.elf file and pass that to the script?
Susan

AussieSusan, 2020-03-05:
OK- I thinkI have your instructions worked out.

My (possibly updated) backtrace is:

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40152bb8  PS      : 0x00060230  A0      : 0x8015a739  A1      : 0x3ffd2aa0  
A2      : 0x3ffc6198  A3      : 0x00000000  A4      : 0x3f425fe1  A5      : 0x00000003  
A6      : 0x3ffc29d4  A7      : 0x3ffc29e0  A8      : 0x00000000  A9      : 0x3ffd2a60  
A10     : 0x3ffc6198  A11     : 0x00000000  A12     : 0x00000004  A13     : 0x3ffd2aa0  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40152bb5:0x3ffd2aa0 0x4015a736:0x3ffd2ad0 0x4015a1f9:0x3ffd2b10 0x401551aa:0x3ffd2b40 0x40153ca5:0x3ffd2b70 0x40153cb7:0x3ffd2b90 0x40158d6a:0x3ffd2bb0 0x400f7b8f:0x3ffd2bd0 0x40098dd5:0x3ffd2bf0

and the output from the backtrace script is:

BT-0: 0x40152bb5 is in ble_gattc_rx_read_type_complete (/users/susan/micropython/ports/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:4543).
4543	        rc = rx_entry->cb(proc, status);
BT-1: 0x4015a736 is in ble_att_clt_rx_read_type (/users/susan/micropython/ports/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_clt.c:454).
454	    ble_gattc_rx_read_type_complete(conn_handle, rc);
BT-2: 0x4015a1f9 is in ble_att_rx (/users/susan/micropython/ports/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att.c:503).
503	    rc = entry->bde_fn(conn_handle, om);
BT-3: 0x401551aa is in ble_hs_hci_evt_acl_process (/users/susan/micropython/ports/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:870).
870	        rc = rx_cb(conn->bhc_rx_chan);
BT-4: 0x40153ca5 is in ble_hs_process_rx_data_queue (/users/susan/micropython/ports/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:238).
238	        ble_hs_hci_evt_acl_process(om);
BT-5: 0x40153cb7 is in ble_hs_event_rx_data (/users/susan/micropython/ports/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:539).
539	    ble_hs_process_rx_data_queue();
BT-6: 0x40158d6a is in nimble_port_run (/users/susan/micropython/ports/esp32/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:121).
121	    ev->fn(ev);
BT-7: 0x400f7b8f is in ble_host_task (nimble.c:38).
38	    nimble_port_run();  //This function will return only when nimble_port_stop() is executed.

Susan

dpgeorge, 2020-03-11:
Should hopefully be fixed by bd746a46309efc261d6124b546b5bf6775d47460 and possibly 1937fb22abc8e6192f6acfc037f2e9fd59fc6d8c

Reopen if the issue still persists.

AussieSusan, 2020-03-12:
Still crashes using a normal build and a DEBUG=1 build.

The backtrace is:

reading backtrace from stdin
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40152d60  PS      : 0x00060530  A0      : 0x8015a8e1  A1      : 0x3ffd1620  
A2      : 0x3ffc6190  A3      : 0x00000000  A4      : 0x3f425fe1  A5      : 0x00000003  
A6      : 0x3ffc29cc  A7      : 0x3ffc29d8  A8      : 0x00000000  A9      : 0x3ffd15e0  
A10     : 0x3ffc6190  A11     : 0x00000000  A12     : 0x00000004  A13     : 0x3ffd1620  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40152d5d:0x3ffd1620 0x4015a8de:0x3ffd1650 0x4015a3a1:0x3ffd1690 0x40155352:0x3ffd16c0 0x40153e4d:0x3ffd16f0 0x40153e5f:0x3ffd1710 0x40158f12:0x3ffd1730 0x400f7c27:0x3ffd1750 0x40098dd5:0x3ffd1770

Rebooting...
BT-0: 0x40152d5d is in ble_gattc_rx_read_type_complete (/Users/Susan/micropython/ports/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:4543).
4543	        rc = rx_entry->cb(proc, status);
BT-1: 0x4015a8de is in ble_att_clt_rx_read_type (/Users/Susan/micropython/ports/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_clt.c:454).
454	    ble_gattc_rx_read_type_complete(conn_handle, rc);
BT-2: 0x4015a3a1 is in ble_att_rx (/Users/Susan/micropython/ports/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att.c:503).
503	    rc = entry->bde_fn(conn_handle, om);
BT-3: 0x40155352 is in ble_hs_hci_evt_acl_process (/Users/Susan/micropython/ports/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:870).
870	        rc = rx_cb(conn->bhc_rx_chan);
BT-4: 0x40153e4d is in ble_hs_process_rx_data_queue (/Users/Susan/micropython/ports/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:238).
238	        ble_hs_hci_evt_acl_process(om);
BT-5: 0x40153e5f is in ble_hs_event_rx_data (/Users/Susan/micropython/ports/esp32/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:539).
539	    ble_hs_process_rx_data_queue();
BT-6: 0x40158f12 is in nimble_port_run (/Users/Susan/micropython/ports/esp32/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:121).
121	    ev->fn(ev);
BT-7: 0x400f7c27 is in ble_host_task (nimble.c:38).
38	    nimble_port_run();  //This function will return only when nimble_port_stop() is executed.

Susan

JohannesNeu, 2020-03-12:
@AussieSusan I have been trying to compile esp32 micropython for a week and used this guide https://www.microdev.it/wp/en/2018/08/08/esp32-micropython-compiling-for-esp32/ on  ubuntu 18.04 and ubuntu 16.04 in a VM but I always get a pyparsing version error (>2.4) although pip shows that 2.3.1 is installed. Can you show me how you did it, so I can take a trace as well?

AussieSusan, 2020-03-13:
Off Topic (sort of)
I started with https://github.com/micropython/micropython to get the MicroPython source. I followed the instructions to (and including) the mpy-cross part.
I then followed the instructions at https://github.com/micropython/micropython/tree/master/ports/esp32 and followed the esp-idf installation instructions for my iMac. There is one small 'trick': after you clone the esp-idf.git, you need to cd into the esp-idf folder before you run the git checkout command. (Run make before you do this to get the correct hash for the checkout.)
Also I created a GNUmakefile as suggested (MacOS is case insensitive) and also had to edit the 'Makefile' to remove the reference to 'python2' as 'python' defaults to Python 3 on my system - and there is no 'python2' (anymore).
There are a few extra things in the GNUmakefile about the board, port and flash_mode but I don't think they are relevant here.
After all that is done, then running 'make DEBUG=1' followed by the erase and deploy all worked.
I hope this helps... even if it is on a Mac and not Ubuntu

agners, 2020-04-12:
I am trying to connect to Xiaomi Mi Flora devices as well, and seem to experience a similar/the same crash:

```
Device found: 0 b'\x80\xea\xca\x88\xe0\x0f' True -31 b'\x02\x01\x06\x03\x02\x95\xfe\x14\x16\x95\xfeq \x98\x00\x85\x0f\xe0\x88\xca\xea\x80\r\x04\x10\x02\x1b\x01'
Services: [UUID16(0xfe95)]
UUID16(0xfe95)
Found sensor: 0 b'\x80\xea\xca\x88\xe0\x0f' ?
Connecting... 0 b'\x80\xea\xca\x88\xe0\x0f'
GAP procedure initiated: connect; peer_addr_type=0 peer_addr=80:ea:ca:88:e0:0f scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=16 max_ce_len=768 own_addr_type=0
Done
GATT procedure initiated: discover all services
GATT procedure initiated: discover all characteristics; start_handle=16 end_handle=34
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4015c4f4  PS      : 0x00060230  A0      : 0x80164065  A1      : 0x3ffbb460  
A2      : 0x3ffc6fcc  A3      : 0x3ffbb4a0  A4      : 0x3f428fc2  A5      : 0x3ffc2d38  
A6      : 0x00000003  A7      : 0x3ffc2d0c  A8      : 0x00000000  A9      : 0x3ffbb420  
A10     : 0x3ffc6fcc  A11     : 0x3ffbb4a0  A12     : 0x00000004  A13     : 0x3ffbb460  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4009c484  LEND    : 0x4009c4b2  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x4015c4f1:0x3ffbb460 0x40164062:0x3ffbb490 0x4015e699:0x3ffbb4d0 0x40160f89:0x3ffbb500 0x4015e2e1:0x3ffbb530 0x4015e2f3:0x3ffbb550 0x4015d182:0x3ffbb570 0x400f9317:0x3ffbb590 0x400999f9:0x3ffbb5b0
         
Rebooting...
```

This is reproducible with the `tinypico-idf4-20200316-v1.12.bin` build as well as the latest master build `tinypico-idf4-20200412-v1.12-357-g740946736.bin`.

Also tried idf3 builds, but the same there too.

I can connect to the same device from a ESP32 using Arduino, which from what I understand uses the "regular"  ESP-IDF BLE API instead of the NimBLE API. So this supports the assumption that this is related to how the NimBLE API is used.

agners, 2020-04-12:
With the hints in https://github.com/espressif/esp-idf/issues/4913 I got communication with Mi Flora devices half way working. But it required quite some "hackarounds": I basically sleep some time after service and characteristics discovery before continuing. It seems that the NumBLE implementation on ESP32 does not allow to directly trigger characteristic searches when services have been found, and no reads when characteristics have been found. I am wondering if this is specific to ESP32 or if the regular NumBLE has this limitation as well... What would be helpful is if MicroPython would include `status` in `_IRQ_GATTC_SERVICE_RESULT` and also trigger the irq when status is `BLE_HS_EDONE`.

This is the MicroPython code I have currently. It works ~1 out of 5 times on my desk :astonished: : https://github.com/agners/micropython-miflora

codefreax, 2020-04-13:
Somewhat related: I posted PR https://github.com/micropython/micropython/pull/5906 that exposes the completion of gattc scans (NimBLE's BLE_HS_EDONE status) as a separate event.

JohannesNeu, 2020-06-05:
I have seen that the changes to show when the discovery has ended have been merged ( #5906  ), does this mean we can test with a new version of micropython?

jimmo, 2020-06-09:
> I have seen that the changes to show when the discovery has ended have been merged ( #5906 ), does this mean we can test with a new version of micropython?

Yes, the latest nightly builds will have these new events. You'll need to make some modifications to your code:
 - Update the event codes (see https://docs.micropython.org/en/latest/library/ubluetooth.html#ubluetooth.BLE.irq for the new list and updated values)
 - Remove the `trigger` arg from ble.irq()
 - Listen for the *_DONE events.

TimonPeng, 2020-06-09:
Crash too.

```python
import bluetooth

ble = bluetooth.BLE()
ble.gap_scan(0, 1000000, 500000)
```

log:

```
assertion "mu->handle" failed: file "/home/micropython/esp-idf-v4/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c", line 185, function: npl_freertos_mutex_pend
abort() was called at PC 0x4015e58f on core 1
```

using `esp32-idf4-20191220-v1.12.bin`

JohannesNeu, 2020-06-09:
@TimonPeng that is the old stable version, I am trying https://micropython.org/resources/firmware/esp32-idf3-20200609-unstable-v1.12-516-g29e258611.bin right now which was built today.

TimonPeng, 2020-06-09:
@JohannesNeu thank you! but not works for my board:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
```

BTW, why idf3 don't exist bluetooth module? idf3 is support for ble too.

jimmo, 2020-06-10:
> OSError: [Errno 19] ENODEV

you need

`ble.active(True)`

before the call to `ble.gap_scan()`

to enable the Bluetooth peripheral and NimBLE stack.

> BTW, why idf3 don't exist bluetooth module? idf3 is support for ble too.

Although IDF3 has supported BLE for a long time, we specifically use the NimBLE stack (rather than bluedroid). NimBLE wasn't backported to IDF3 until after we first released support for BLE.
However, it is now available on IDF3 builds, so any of the IDF3 "unstable" (i.e. post 1.12) versions on http://micropython.org/download/esp32/ will include BLE.

TimonPeng, 2020-06-10:
@jimmo learned! thank you!

JohannesNeu, 2020-06-11:
@jimmo 
I changed the code a bit and micropython prints several addresses to the screen but only ever adds the same addr to my list. Am I missing some IRQ concept?
```
    elif event == _IRQ_SCAN_RESULT:
        # A single scan result.
        addr_type, addr, adv_type, rssi, adv_data = data
        print(addr)
        list_of_devices.append(addr)
```
The printout is the following:
```
>>> import main
b',\xab3\x01,\x89'
b'p*\xd5z\x8a\xce'
b',\xab3\x01,\x89'
b',\xab3\x01,\x89'
b',\xab3\x01,\x89'
b'p*\xd5z\x8a\xce'
b'?$^\xacw\x10'
b'?$^\xacw\x10'
b'p*\xd5z\x8a\xce'
b'?$^\xacw\x10'
b'?$^\xacw\x10'
b',\xab3\x01,\x89'
b'p*\xd5z\x8a\xce'
b',\xab3\x01,\x89'
b'(\xff<\xb5#.'
b',\xab3\x01,\x89'
b'(\xff<\xb5#.'
b',\xab3\x01,\x89'
b'p*\xd5z\x8a\xce'
b',\xab3\x01,\x89'
b',\xab3\x01,\x89'
b',\xab3\x01,\x89'
b',\xab3\x01,\x89'
b',\xab3\x01,\x89'
b'p*\xd5z\x8a\xce'
b'p*\xd5z\x8a\xce'
b',\xab3\x01,\x89'
[b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89', b',\xab3\x01,\x89']
```
For every print executed from within the bt_irq function due to the elif mentioned above I will see different addresses. But when I print the whole list: there is just one addr over and over again. I counted the amount of addresses and bot times it is 27, which means for every found device I am appending the same address to my list, although the addr is actually different. I am a bit confused.

friebel, 2020-06-11:
> For every print executed from within the bt_irq function due to the elif mentioned above I will see different addresses. But when I print the whole list: there is just one addr over and over again.

I ran into the same issue when starting to use mpy ble. From http://docs.micropython.org/en/latest/library/ubluetooth.html:

> Note: the addr, adv_data, char_data, notify_data, and uuid entries in the tuples are references to data managed by the ubluetooth module (i.e. the same instance will be re-used across multiple calls to the event handler). If your program wants to use this data outside of the handler, then it must copy them first, e.g. by using bytes(addr) or bluetooth.UUID(uuid).

jimmo, 2020-06-11:
Yep, exactly what @friebel said. It's done this way so that the IRQ handler doesn't have to do any heap allocation (which would create more fragmentation). The idea is that you only copy the data if it's a device you care about.

ElreboCM, 2020-07-07:
I am also receiving a kernel when calling gattc_discover_descriptors inside the handler.
'
if conn_handle == self._conn_handle:
self._value_handle.append(value_handle)
if (len(self._value_handle) == 3):
self._ble.gattc_discover_descriptors(self._conn_handle, self._start_handle, self._end_handle)`

**Error:**
`GATT procedure initiated: discover all descriptors; chr_val_handle=10 end_handle=22
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40151b34  PS      : 0x00060a30  A0      : 0x80158bdb  A1      : 0x3ffbb530  
A2      : 0x3ffc6e2c  A3      : 0x00000000  A4      : 0x3f4286fe  A5      : 0x3ffc2b94  
A6      : 0x00000003  A7      : 0x3ffc2b68  A8      : 0x00000000  A9      : 0x3ffbb4f0  
A10     : 0x3ffc6e2c  A11     : 0x00000000  A12     : 0x00000004  A13     : 0x3ffbb530  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000 `

ElreboCM, 2020-07-07:
I am receiving also a kernel panic when using the latest build (esp32-idf4-20200707-unstable-v1.12-614-gc2317a3a8) for esp32 (no-spiram).

This the call:

```def read(self, callback):
        if not self.is_connected():
            return
        self._read_callback = callback
        self._ble.gattc_read(self._conn_handle, self._value_handle[str(_MY_UUID)])`

and this the error:
```
GATT procedure initiated: read; att_handle=12
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40154de4  PS      : 0x00060c30  A0      : 0x8015e3ff  A1      : 0x3ffd1240  
A2      : 0x3ffc50c0  A3      : 0x00000000  A4      : 0x3f42bb27  A5      : 0x3f40286e  
A6      : 0x3f431642  A7      : 0x3f4316a8  A8      : 0x00000000  A9      : 0x3ffd1200  
A10     : 0x3ffc50c0  A11     : 0x00000000  A12     : 0x00000004  A13     : 0x3ffd1240  
A14     : 0x00000003  A15     : 0x3ffc32a4  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x40154de1:0x3ffd1240 0x4015e3fc:0x3ffd1270 0x4015dec5:0x3ffd12b0 0x401576b9:0x3ffd12e0 0x40155f95:0x3ffd1310 0x40155fa7:0x3ffd1330 0x4015c97e:0x3ffd1350 0x400f7e9f:0x3ffd1370 0x4009b981:0x3ffd1390
```

AussieSusan, 2020-07-08:
With all of these, the key value to look for is EXCVADDR - the virtual address that caused the exception. If it is something like 0x00000004 then it is a fair bet that this is an offset from a NULL pointer.
If you use the traceback information and find the code where the error occurred, it may well have a function call (perhaps BLE_GATTC_RX_EXTRACT_RX_ENTRY) that returns a pointer (such as 'rx_entry') that shortly after is used to call a 'callback' function. [See one of the early posts in this thread for an example.]
It so happens that the callback address is at offset 0x00000004 in the struct being pointed to. However if the pointer is NULL then it will try to access the low address referenced in EXCVADDR.
Until the original source code gets fixed, all you can really do is to test the pointer and only use it is not NULL. Of course this only fixes the symptom - there is probably something causing the return of a NULL pointer deeper in the code but it seems to be written in a somewhat 'impenetrable' style (my opinion only!),
That, and continue to complain to the original code authors that there is something wrong with their code that they need to fix!
Susan

laservert, 2020-10-12:
I tried to test https://github.com/agners/micropython-miflora

In (event == _IRQ_GATTC_CHARACTERISTIC_RESULT), it seems "self._characteristics_handle[uuid] = value_handle" (line 181) does not work.

Curiously, I obtain:
First discovery, self._characteristics_handle  = {UUID128('00001a00-0000-1000-8000-00805f9b34fb'): 51}

Second discovery, self._characteristics_handle = {UUID128('00001a01-0000-1000-8000-00805f9b34fb'): 51, UUID128('00001a01-0000-1000-8000-00805f9b34fb'): 53}

Third discovery, self._characteristics_handle  = {UUID128('00001a02-0000-1000-8000-00805f9b34fb'): 56, UUID128('00001a02-0000-1000-8000-00805f9b34fb'): 53}

The dictionary is not correctly actualized by Micropython! But I can't propose a new issue...