micropython/micropython #5687: Dirty flashing or upgrading from a different firmware does not work

mordae, 2020-02-22:
Hello. I have flashed Tasmota on a couple of Tuya bulbs using `tuya-convert` and now I would like to flash MicroPython and give the LEDs to a bunch of kids I teach Python to play with. The problem is that I cannot exactly wipe the bulbs before flashing them without opening them up and that is something I would like to avoid since I don't have a ~230V license. And I am afraid that without wiping them beforehand MicroPython won't boot up.

Why? I have tested the Tasmota->MicroPython migration path on a Sonoff esp8285 board and after figuring out `SetOption78 1` on Tasmota and flashing MicroPython I get only a never-ending stream of random garbage on UART. Same thing happens when I dirty-flash MicroPython using `esptool.py`. Everything works fine when I erase the flash beforehand.

I don't know if it helps, but when I manually flash 0x0fc000 it starts to work:

```bash
esptool -b 460800 write_flash --flash_mode=dout 0x0fc000 esp_init_data_default.bin
```

Please advise.

peterhinch, 2020-02-23:
When problems occur installing MicroPython, the first advice is to erase flash. I would therefore be unwilling to install MicroPython on any device unless the option to erase was available.

As a general point, queries like this should be raised in [the forum](http://forum.micropython.org/index.php) and not in this bug tracker.

mordae, 2020-02-24:
But then the kids won't be able to brag about Python-programming their bedroom light bulbs. :-)

I believe the problem lies with `firmware_start`, that only writes the `default_init_data` when the destination is empty. Tasmota dirties it. Surprisingly, it seems that `firmware_start` does not work even when I make it unconditional. I will order an esp8266 board to double check.

closed 2024-10-01 by mordae