micropython/micropython #5693: [mboot] It will stay in the dfu mode forever and not reset

rolandvs, 2020-02-24:
When using `mboot` to replace the code with `dfu` commands all works fine. However, `mboot` stays in `dfu` mode and not resetting the device. This is tested using  `mboot` with the `H743`.

andrewleech, 2020-02-25:
Is this with a released version of micropython/mboot or current master?
Are you using the pydfu script included in the micropython/tools folder?  
At the end of flashing the dfu file does it print out `Exiting DFU...`?

The mechanism to exit mboot / reboot to application is a separate dfu command, depending on the tool being used this may not be sent automatically. pydfu should send this however.

rolandvs, 2020-02-25:
I tried again using and put in a bit more detail (I should have done that the first time :-)
```
$ make BOARD=NUCLEO_H743ZI2 USE_MBOOT=1 USE_PYDFU=1 deploy
```
I notices some strange behaviour:
- when using PA0 (as defined in mpconfigboard.h) and NRST to put the device into MBOOT (LED is flashing) the DFU mode is not set, I cannot program a new image.
- when booted, using machine.bootloader() puts the device in MBOOT, DFU mode is working, device does not leave this mode (as reported). I have to hit reset.
- when using the BOOT0 - NRST (build-in DFU boot loader from STM) I can load the software with the correct offset and the device leaves DFU mode.

andrewleech, 2020-02-25:
With the first issue, there's to separate pin based methods to get into mboot. 
1. The user switch (c13) can be held down and boot modes will cycle through on the LEDs, cv you then need to let go of the switch at the mode you want.
2. MBOOT_BOOTPIN_PIN can be defined to a pin, then it will be used immediately to jump into mboot. I can't see this out PA0 defined in the current mpconfigboard for that nucleo, have you added it yourself?
I use MBOOT_BOOTPIN_PIN on my custom board where it works as expected.