micropython/micropython #5703: [esp32] when booting, repl display wrong version?

ccccmagicboy, 2020-02-27:
hi there, my daily build display **MicroPython v1.11-888-g3f1b81009** when booting with repl, but the source code itself is a **v1.12 implementation**, is there something mistake?
```
I (633) psram: This chip is ESP32-D0WD
I (634) spiram: Found 64MBit SPI RAM device
I (634) spiram: SPI RAM mode: flash 40m sram 40m
I (637) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (644) cpu_start: Pro cpu up.
I (648) cpu_start: Application information:
I (652) cpu_start: Compile time:     Feb 27 2020 11:11:50
I (659) cpu_start: ELF file SHA256:  0000000000000000...
I (664) cpu_start: ESP-IDF:          v3.3.1
I (669) cpu_start: Starting app cpu, entry point is 0x40084078
I (0) cpu_start: App cpu up.
I (1545) spiram: SPI SRAM memory test OK
I (1545) heap_init: Initializing. RAM available for dynamic allocation:
I (1546) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1552) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1558) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1564) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (1570) heap_init: At 3FFCCAC8 len 00013538 (77 KiB): DRAM
I (1576) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1583) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1589) heap_init: At 4009DEE4 len 0000211C (8 KiB): IRAM
I (1596) cpu_start: Pro cpu start user code
I (272) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (150) pm_esp32: Frequency switching config: CPU_MAX: 240, APB_MAX: 240, APB_MIN: 240, Light sleep: DISABLED
MicroPython v1.11-888-g3f1b81009 on 2020-02-27; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> import sys
>>> sys.
__class__       __name__        argv            byteorder
exit            implementation  maxsize         modules
path            platform        print_exception
stderr          stdin           stdout          version
version_info
>>> sys.platform()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'str' object isn't callable
>>> sys.platform
'esp32'
>>> sys.version
version         version_info
>>> sys.version
version         version_info
>>> sys.version
'3.4.0'
>>> sys.version_info
(3, 4, 0)
>>> sys.implementation
(name='micropython', version=(1, 12, 0), mpy=10757)
>>> 
>>> 
>>> 
>>> 
>>> 
>>> 
```

jimmo, 2020-02-27:
This suggests your repo is in a state where it doesn't see the v1.12 tag in the history reachable from the current HEAD.

What do you get if you run
```
git describe --dirty --always
```

The likely cause is that you've merged or rebased on top of (upstream) master, so you have the changes from 1.12 (and onwards) but your history is rewritten (i.e. the v1.12 tag points somewhere else).

If you want to get back to a clean state (warning, might destroy any local changes), assuming that the "origin" remote is upstream (i.e. github.com:micropython/micropython), check with `git remote -v` first.

```
# Make a branch that points at whatever you have now.
git checkout -b my-current-state

# Update origin
git fetch origin

# Go to the master branch, and make it exactly identical to upstream.
git checkout master
git reset --hard origin/master

# Should say "v1.12...."
git describe --dirty --always
```

Then you can still go back to my-current-state if you want to access any commits you made, but at least master will be up to date.

ccccmagicboy, 2020-02-27:
Thanks a lot. :) Indeed, what I do is apply a old PR, and rebase it at master in my repo. Now trying to use cherry-pick to build my firmware.

closed 2020-02-27 by ccccmagicboy