micropython/micropython #5705: ESP32: WPA2-Enterprise support

wgaylord, 2020-02-27:
Hello, I am making this issue to request support for WPA2 enterprise network that use EAP.

I had late 2019 posted some code on how to do so that I used in a custom build of micropython so that someone with more knowledge of how to properly add it could take a look.

https://forum.micropython.org/viewtopic.php?f=18&t=7219

Looks like no one ever did which is why I am creating this issue to bring attention to this.

wgaylord, 2020-09-15:
Will there ever be the possibility of someone looking into this? I don't want to keep building custom firmware.  

Or should I spend my time trying to figure out the proper way to implement this in and make a PR? (Wish there was docs about how to do the exceptions, checks and stuff on the C side of things)

tve, 2020-09-16:
Yup, a pull request is the way these things happen. I wouldn't even have a WPA2 enterprise AP to test against...
WRT handling errors, there's a `check_esp_err` function (https://github.com/micropython/micropython/blob/master/ports/esp32/mphalport.c#L57-L87) used for example in https://github.com/micropython/micropython/blob/master/ports/esp32/esp32_partition.c#L154

GC-RnD, 2020-10-07:
We have a project in development that is now required to have WPA2-Enterprise encryption.
I see the Arduino and the Pycom groups have this ability.
Has anyone ever seen a library available for this.
This is becoming required for commercial networks and I hope the micropython group will move forward to including Enterprise connectivity.

wgaylord, 2020-12-14:
I will be trying to make a PR for this sometime soon. Have to set up my Micropython build system again. Plus I am probably going to be loosing access the the WPA2 EAP network I needed this for soon because I am graduating in a few days.

wgaylord, 2021-05-21:
I will be working on a PR for this. Finally get a good amount of experience working with Micropython's internals that I think i can make a PR properly.

ATM-HSW, 2021-11-23:
Anything news about implementing WPA2 EAP?

MATTYGILO, 2022-05-16:
How's this coming on?

one0410, 2023-03-04:
Hi, I really need WPA2-Enterprise feature on ESP32.
Please add this feature.
Thx.

SL2021-Dev, 2023-08-31:
Same...
I also have been using hacks with help from other members in order to provide PEAP .
It has been some time now and an offical release should be worked on.
I have also mentioned this in the Core Development forum....
https://github.com/orgs/micropython/discussions/12143

Please... seriously consider adding this to Micropython !!!

mattytrentini, 2025-02-20:
For those folks interested in this feature, PR #16463 provides an implementation. If you can provide a review or, better yet, test the PR that would be very helpful!