micropython/micropython #5714: ESP8266 Pin IRQ regression

peterhinch, 2020-03-02:
This [device driver for IR remotes](https://github.com/peterhinch/micropython-async/blob/master/nec_ir/aremote.py) formerly worked on ESP8266. It now crashes with no changes to code. The problem is related to hard Pin IRQ's and the following code sample demonstrates it.
```python
from machine import Pin
from time import sleep
import gc

pin = Pin(13, Pin.IN)
count = 0
def cb_pin(_):
    global count
    count += 1

ei = pin.irq(handler = cb_pin, trigger = (Pin.IRQ_FALLING | Pin.IRQ_RISING), hard = True)
# ei = pin.irq(handler = cb_pin, trigger = Pin.IRQ_FALLING, hard = True)

while True:
    sleep(1)
    print(count)
    gc.collect()
```
To test the pin is fed with a burst of 14 pulses of Manchester encoded data repeating every 104ms. Pulses are 889μs or 1778μs wide, the burst lasting 24.9ms. A crash with reboot occurs usually within a few seconds.

A crash also occurs if the commented-out code is used, ensuring that the minimum time between triggers is a leisurely 1.8ms. The crash takes longer to trigger.

Testing was with firmware built from source acquired today.

dpgeorge, 2020-03-03:
I can't reproduce this.  I tried with latest esp8266 build and applied a 20Hz square wave to Pin(13) (from a pyboard).  I have all network interfaces disabled.

What is the crash output?

peterhinch, 2020-03-03:
This was fixed by erasing flash. This is new to me: previously when an erase was required the system was evidently broken. This time it ran substantial programs but fell over on an IRQ.

Apologies for the incorrect issue.

dpgeorge, 2020-03-03:
> This time it ran substantial programs but fell over on an IRQ.

That is quite strange...

> Apologies for the incorrect issue.

No problem!

peterhinch, 2020-03-03:
Oh, not so simple. It turns out the fault only occurs if there is a station interface open.

When I erased flash the network connection was lost so I got the same (good) result as you. If I run my connect script the failure is back. There is no network activity, merely an existing connection.

I can't post the outcome of the crash as every time it occurs my bash/rshell screen is cleared losing all history.

dpgeorge, 2020-03-03:
Ok, I can reproduce a crash.  I connect Pin(13) to Pin(2) and put a PWM signal on Pin(2) with `machine.PWM(machine.Pin(2), freq=500, duty=50)`.  When I ping the esp8266 from a PC everything works fine and `count` increases at 1kHz (due to falling and rising edge being counted).  But when I stop the pings, after a few seconds the esp8266 prints "pm 572" and then stops responding completely.  My guess is that it's related to wifi power management.

peterhinch, 2020-03-03:
Interesting. That casts doubt on my claim of a regression.

The [device driver](https://github.com/peterhinch/micropython-async/blob/master/nec_ir/aremote.py) in my OP has run in a uasyncio application using MQTT. WiFi was actively in use.

I have a stand-alone test script for the driver. It's possible that when I ran it on ESP8266 no connection was established.

peterhinch, 2020-04-22:
This bug is a total pig. It's caught me out again with I2C use suffering rare bus timeouts unless the board is repeatedly pinged.

It's a clear regression: code tested thoroughly two years ago now fails at rare intervals. See [forum thread](https://forum.micropython.org/viewtopic.php?f=16&t=8207&#p46759).

dpgeorge, 2020-05-02:
Should be properly fixed now.

commit 0bd58a56 (2020-05-02): esp8266/machine_pin: Move pin_intr_handler to iRAM, de-support hard IRQ.

GPIO interrupts can occur when the flash ROM cache is in use and so the
GPIO interrupt handler must be in iRAM.  This commit moves the handler to
iRAM, and also moves mp_sched_schedule to iRAM which is called by
pin_intr_handler.

As part of this fix the Pin class can no longer support hard=True in the
Pin.irq() method, because the VM and runtime are too big to put in iRAM.

Fixes #5714.
PR #5962 (2020-05-02): esp8266: put pin interrupt handler in iRAM
Attempt to fix #5714
closed 2020-05-02 by dpgeorge