micropython/micropython #5717: mpy does not support a module/file named in Chinese？

laowu2019, 2020-03-03:
I wrote a .py file with a chinese name and tried to import it, but failed. 

Is this not implemented?

Functions/Classes/args with chinese names are OK.

dpgeorge, 2020-03-04:
This should be supported, but it will depend on how the filesystem stores non-ASCII chars, and how the file is named in the import.

Eg, with the following file called `αβψδϵφ.py`:
```
print('αβψδϵφ.py imported')
```
And then another file called `test.py`:
```
import αβψδϵφ
```
I get the following:
```
$ micropython test.py 
αβψδϵφ.py imported
```

laowu2019, 2020-03-04:
YES, indeed. It is filesys related. 

I changed the lang setting of my system to Unicode and it worked.

Thanks.

closed 2020-03-04 by laowu2019