micropython/micropython #5732: Error enabling NUS service on NRF52840

skateone, 2020-03-06:
hi there.
i am trying to compile micropython with NUS service enabled.
I get the below compilation erorr:

drivers/bluetooth/ble_uart.c: In function 'gatts_event_handler':
drivers/bluetooth/ble_uart.c:163:21: error: implicit declaration of function 'mp_keyboard_interrupt' [-Werror=implicit-function-declaration]
  163 |                     mp_keyboard_interrupt();
      |                     ^~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors

i have added a custom board but its using the NRF52840 chipset and S140 softdevice.

Deenis, 2020-06-25:
Having the same issue.

Deenis, 2020-06-25:
I added the missing includes, in the ble_uart.c

`#include "py/mphal.h"
#include "py/runtime.h"`

Which solved the problem for me.

glennrub, 2020-06-25:
@Deenis , I've pushed a set of fixes now. It should resolve this issue, but also a few others. You can try them out if you like.

dpgeorge, 2020-07-01:
Should be fixed by 9dfb4ae6aacd5eb4317f7a474b9da00c1cec9088

closed 2020-07-01 by dpgeorge