micropython/micropython #5733: machine.RTC.init documentation is incorrect

dhylands, 2020-03-07:
The machine.RTC documentation has some discrepancies.

The ESP8266 machine.RTC modules doesn't appear to have an init function, but rather has a datetime function.
https://github.com/micropython/micropython/blob/8db5d2d1f13aa6413be11bc13c94df72296a0070/ports/esp8266/machine_rtc.c#L127
The order of the arguments to datetime appears to be (year, month, day, wday, hour, minute, seconds, milliseconds)

The ESP32 machine.RTC module
https://github.com/micropython/micropython/blob/8db5d2d1f13aa6413be11bc13c94df72296a0070/ports/esp32/machine_rtc.c#L87
has an init function and a datetime function. The order of the arguments for datetime and init appears to be (year, month, day, hour, minute, seconds, microseconds)

The stm32 machine.RTC module forwards to pyb.RTC
https://github.com/micropython/micropython/blob/8db5d2d1f13aa6413be11bc13c94df72296a0070/ports/stm32/rtc.c#L515
which has an init function and datetime function, but the init function doesn't take any srguments. The datetime function appears to use the order (year, month, day, wday, hour, month, seconds, microseconds)

The documentation documents an init function with arguments in a different order to any of the above and doesnt't mention a datetime function.

dhylands, 2020-03-07:
I think that the documentation should be updated to reflect the code, but I thought I would raise this issue to discuss that before submitting a PR.

peterhinch, 2020-03-08:
See https://github.com/micropython/micropython/pull/5553

robert-hh, 2021-05-20:
Just came along the same mess while implementing a machine.RTC module for mimxrt. It seem the only posrt using rtc.now() is the cc3200 one, which is stale. So I went for rtc.datetime().

evilhamsterman, 2022-10-18:
This is still messed up. The Raspberry Pi Pico doesn't have a init() function, also the wday in `RTC.datetime((year, month, day, wday, hour, month, seconds, microseconds))` seems unneeded, it isn't used in any other place and it is something that can be calculated)

peterhinch, 2022-10-18:
You don't need `wday` when setting time but it is useful when reading it, and it makes sense to use the same tuple for read and write.