micropython/micropython #5737: ESP32 Confusing behaviour of PWM after soft reset

peterhinch, 2020-03-08:
Consider:
```python
from machine import Pin, PWM
pin = Pin(23, Pin.OUT)
pwm = PWM(pin, freq=38000, duty=338)
```
This works as expected. If you issue **ctrl-d** the waveform on the pin continues unchanged. If you paste the above code again, you cause a low output with
```python
pwm.duty(0)
```
Subsequently changing the duty value to a value > 0 has no effect. This is surprising.

There is a simple workround of issuing **pwm.deinit()** at the start. The current state could be confusing to newcomers. It should either be documented or changed so that a soft reset de-initialises any PWM instance.

tve, 2020-03-08:
Isn't this an issue with almost all peripherals on the ESP32 because a soft reset doesn't reset either the hardware nor the ESP-IDF state?

peterhinch, 2020-03-08:
I have little experience of programming ESP32 at pin level so I can't answer that. Documenting it with a code sample might suffice:
```python
from machine import Pin, PWM
pin = Pin(23, Pin.OUT)
pwm = PWM(pin)
pwm.deinit()
pwm.init(freq=38000, duty=338)
```
This reinstates control after a soft reset; users may still be surprised that the waveform continues after **ctrl-d**. On the other hand, timer callback programming on the Pyboard has similar behaviour.

tve, 2020-03-08:
More info always helps, but I think the real doc needs to go into the soft reset to say that the effect of a soft reset on peripherals on the esp32 is unpredictable.

MrSurly, 2020-03-08:
Correct

jej, 2021-01-26:
I got this problem today after switching from esp8266 to esp32. I didn't have this problem on 8266... thanks to Peter for the workaround.

IhorNehrutsa, 2022-01-14:
Solved in latest #8075.
Please test and close.

gschwaer, 2023-09-17:
A similar issue occurs on the ESP8266 when using GPIO2 with PWM. It's a bit different in that the waveform is also stopped (I assume since GPIO2 is a boot select pin). So doing:
```python
pwm2 = machine.PWM(machine.Pin(2), freq=500, duty=512)  # waveform appears
# CTRL-D: waveform vanishes
pwm2 = machine.PWM(machine.Pin(2), freq=500, duty=512)  # nothing happens, no waveform, me confused
```
The workaround is the same though:
```python
pwm2 = machine.PWM(machine.Pin(2), freq=500, duty=512)  # nothing happens, no waveform, gimme pwm2 handle
pwm2.deinit()
pwm2 = machine.PWM(machine.Pin(2), freq=500, duty=512)  # waveform appears
```