micropython/micropython #5738: Is ESP32 BLE Advertiser unconnectable status means it's unscannable?

Walkline80, 2020-03-09:
I use 2 esp32 boards as Advertiser and Scanner, here is my sample codes:

Advertiser part:
```python
import ubluetooth as bt

def main():
    ble = bt.BLE()

    ble.active(False)
    ble.active(True)

    # iBeacon payload
    adv_payload = bytearray(b"\x02\x01\x06\x1a\xffL\x00\x02\x15MyA\x9f\xb1\x80O\xf6\x8c\xb6\x9f\xa1\xb5\x7f\xb1h'\x1a'f\xc7")

    ble.gap_advertise(None)
    ble.gap_advertise(100000, adv_data=adv_payload, connectable=False)

if __name__ == "__main__":
    main()
```

Scanner part:
```python
import ubluetooth as bt

def main():
    def irq(event, data):
        if event == 1 << 4: # scan result
            print(data)

    ble = bt.BLE()

    ble.active(False)
    ble.active(True)

    ble.irq(irq)

    ble.gap_scan(None)
    ble.gap_scan(0, 1000000, 100000)

if __name__ == "__main__":
    main()
```

While both boards runs correctly, the Scanner can not find the Advertiser's broadcast, but running `nRF Connect` with my mobile can find it.

I've tried turn the Advertiser's `connectable` param to `True`, and then the Scanner can find it easily.

Is there any solution for this issue?

dpgeorge, 2020-03-09:
> While both boards runs correctly, the Scanner can not find the Advertiser's broadcast, but running `nRF Connect` with my mobile can find it.
> 
> I've tried turn the Advertiser's `connectable` param to `True`, and then the Scanner can find it easily.
> 
> Is there any solution for this issue?

Yes I've noticed this bug as well.  It's because `extmod/nimble/modbluetooth_nimble.c:gap_scan_cb()` is not handling all incoming scan events.

dpgeorge, 2020-03-10:
I posted a fix in #5743.

dpgeorge, 2020-03-11:
Fixed by dd0bc26e65734b8a4fafa3769008e92e2ec6645d

commit dd0bc26e (2020-03-11): extmod/modbluetooth: Change scan result's "connectable" to "adv_type".

This commit changes the BLE _IRQ_SCAN_RESULT data from:

    addr_type, addr, connectable, rssi, adv_data

to:

    addr_type, addr, adv_type, rssi, adv_data

This allows _IRQ_SCAN_RESULT to handle all scan result types (not just
connectable and non-connectable passive scans), and to distinguish between
them using adv_type which is an integer taking values 0x00-0x04 per the BT
specification.

This is a breaking change to the API, albeit a very minor one: the existing
connectable value was a boolean and True now becomes 0x0 [... cut]
PR #5743 (2020-03-11): extmod/modbluetooth: Change scan result's "connectable" to "adv_type".
This PR changes the BLE `_IRQ_SCAN_RESULT` data from:
```
addr_type, addr, connectable, rssi, adv_data
```
to:
```
addr_type, addr, adv_type, rssi, adv_data
```
This allows `_IRQ_SCAN_RESULT` to handle all scan result types (not just connectable and non-connectable passive scans), and to distinguish between them using `adv_type` which is an integer taking values 0x00-0x04 per the BT spec (see change to docs in this PR).

This is a breaking change to the API, albeit a very minor one: the existing `connectable` value was a bo [... cut]
closed 2020-03-11 by dpgeorge