micropython/micropython #5739: PYB UART IRQ/Callbacks

YSP0316, 2020-03-09:
MicroPython v1.12-213-g8db5d2d1f-dirty on 2020-03-08; PYBv1.0 with STM32F405RG.

I always get error “TypeError: extra keyword arguments given” when I use UART.As follows:

>>> u6=pyb.UART(6,115200)
>>> u6.irq(trigger=pyb.UART.any,priority=1,handler=ledtest,wake=pyb.UART.IRQ_RXIDLE)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: extra keyword arguments given

please , Is there any instruction or example reference for pyb.UART.irq?

YSP0316, 2020-03-09:
[image]

robert-hh, 2020-03-09:
It's a documentation issue. For Pyboard, the only "handler", "trigger" and "hard"  are supported as keywords. "priority" and "wake" are defined for the CC3200 port only, which in turn does not have the "hard" keyword.
The fact, that the documentation does not differ between the various ports is truly misleading. It was better before, when the ports were clearly separated in the docs.

peterhinch, 2020-03-09:
@robert-hh Agreed. I am writing cross-platform code and having to resort to experimentation to fill the holes in the docs. I wrote an RFC https://github.com/micropython/micropython/issues/5564 on some issues around portability. There has been no comment from the maintainers - perhaps you'd like to post your views?

tve, 2020-03-09:
@jimmo have you made progress on the docs style sheet to allow per-port hide/show sections?

YSP0316, 2020-03-10:
@

> It's a documentation issue. For Pyboard, the only "handler", "trigger" and "hard" are supported as keywords. "priority" and "wake" are defined for the CC3200 port only, which in turn does not have [...]

so , The correct one is "u6.irq(trigger=pyb.UART.any,handler=ledtest,hard=6)", right?
What is "hard" ? Is the UART Port ?
However, there are still errors, as follows:

[image]

I don't know how to "use uart.irq".

robert-hh, 2020-03-10:
You get the error because pyb.UART.any is a method (telling you if bytes were received), and not an integer, as expected.  The only constant which you might use is pyb.UART.IRQ_RXIDLE. The keyword hard=True/False tells, whether a hard or soft interrupt is done. The latter allows allocating memory. The default value is 'False'.  
In any case, using machine.UART is preferred, since that is the module provided for all ports.

YSP0316, 2020-03-12:
Thank you. I understand.
And I found a case study as follows:

[image]

closed 2020-03-13 by YSP0316