micropython/micropython #5746: ESP32 esp-idf-v3 -> reset

pidou46, 2020-03-10:
Hello,

 calling config('mac') cause reset with latest esp-idf-v3

```
MicroPython` v1.12-233-ga01757670 on 2020-03-10; ESP32 module with ESP32
Type "help()" for more information.
>>> import bluetooth
>>> ble=bluetooth.BLE()
>>> ble.config('mac')
assertion "mu->handle" failed: file "/home/micropython/esp-idf-v3/components/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c", line 185, function: npl_freertos_mutex_pend
abort() was called at PC 0x40167b2b on core 1

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40091bdf:0x3ffd0390 0x40091efd:0x3ffd03b0 0x40167b2b:0x3ffd03d0 0x4015bcfb:0x3ffd0400 0x4015cb58:0x3ffd0420 0x4015cb6f:0x3ffd0440 0x401581e7:0x3ffd0460 0x40102595:0x3ffd0490 0x400f1911:0x3ffd04b0 0x400e3581:0x3ffd04f0 0x400df729:0x3ffd0520 0x400df849:0x3ffd0540 0x400eccd5:0x3ffd0560 0x400e3648:0x3ffd0600 0x400df729:0x3ffd0670 0x400df752:0x3ffd0690 0x401037ab:0x3ffd06b0 0x4010389d:0x3ffd0740 0x400f5391:0x3ffd0770 0x40097331:0x3ffd07a0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5008
ho 0 tail 12 room 4
load:0x40078000,len:10600
ho 0 tail 12 room 4
load:0x40080400,len:5684
entry 0x400806bc
I (531) cpu_start: Pro cpu up.
I (531) cpu_start: Application information:
I (531) cpu_start: Compile time:     Mar 10 2020 12:41:25
I (534) cpu_start: ELF file SHA256:  0000000000000000...
I (540) cpu_start: ESP-IDF:          v3.3.1
I (545) cpu_start: Starting app cpu, entry point is 0x400836ec
I (535) cpu_start: App cpu up.
I (556) heap_init: Initializing. RAM available for dynamic allocation:
I (563) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (569) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (575) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (581) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (587) heap_init: At 3FFCC810 len 000137F0 (77 KiB): DRAM
I (593) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (599) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (606) heap_init: At 40099A80 len 00006580 (25 KiB): IRAM
I (612) cpu_start: Pro cpu start user code
I (295) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
MicroPython v1.12-233-ga01757670 on 2020-03-10; ESP32 module with ESP32
Type "help()" for more information.
```

pidou46, 2020-03-10:
Same thing with esp-idf-v4 with latest firmware
```
MicroPython v1.12-233-ga01757670 on 2020-03-10; ESP32 module with ESP32
Type "help()" for more information.
>>> import bluetooth
>>> ble=bluetooth.BLE()
>>> ble.config('mac')
assertion "mu->handle" failed: file "/home/micropython/esp-idf-v4/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c", line 185, function: npl_freertos_mutex_pend
abort() was called at PC 0x4016135f on core 1

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x4009b000:0x3ffd05d0 0x4009b3e9:0x3ffd05f0 0x4016135f:0x3ffd0610 0x4015572f:0x3ffd0640 0x40156590:0x3ffd0660 0x401565a7:0x3ffd0680 0x40151c1b:0x3ffd06a0 0x401020f1:0x3ffd06d0 0x400f190d:0x3ffd06f0 0x400e3589:0x3ffd0730 0x400df749:0x3ffd0760 0x400df869:0x3ffd0780 0x400eccd1:0x3ffd07a0 0x400e3650:0x3ffd0840 0x400df749:0x3ffd08b0 0x400df772:0x3ffd08d0 0x40103307:0x3ffd08f0 0x401033f9:0x3ffd0980 0x400f5385:0x3ffd09b0 0x4009c091:0x3ffd09e0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5148
load:0x40078000,len:12880
load:0x40080400,len:3484
entry 0x40080630
I (526) cpu_start: Pro cpu up.
I (526) cpu_start: Application information:
I (526) cpu_start: Compile time:     Mar 10 2020 12:49:07
I (529) cpu_start: ELF file SHA256:  0000000000000000...
I (535) cpu_start: ESP-IDF:          v4.0
I (540) cpu_start: Starting app cpu, entry point is 0x40083000
I (532) cpu_start: App cpu up.
I (550) heap_init: Initializing. RAM available for dynamic allocation:
I (557) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (563) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (569) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (575) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (581) heap_init: At 3FFCCA48 len 000135B8 (77 KiB): DRAM
I (588) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (594) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (600) heap_init: At 4009EA08 len 000015F8 (5 KiB): IRAM
I (607) cpu_start: Pro cpu start user code
I (625) spi_flash: detected chip: generic
I (625) spi_flash: flash io: dio
I (626) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
MicroPython v1.12-233-ga01757670 on 2020-03-10; ESP32 module with ESP32
Type "help()" for more information.
```

dpgeorge, 2020-03-11:
Thanks for the report, but I'll close this as a duplicate of #5638 .

dpgeorge, 2020-03-11:
Actually, this is not fixed by the recent 1937fb22abc8e6192f6acfc037f2e9fd59fc6d8c

dpgeorge, 2020-03-11:
To get the MAC, ensure that the BLE is active first: `ble.active(1)`

pidou46, 2020-03-11:
Thank you,
Sorry for disturbance.