micropython/micropython #5755: ESP32 micropython prints 99.0 as 99.00001

goatchurchprime, 2020-03-12:
This is consistent across  ESP32s running the latest build, but does not affect the ESP8266.  

As you can see, the underlying bitwise representation of the integral floating point number is correct (and matches the representation from CPython)-- it's the str() function that goes wrong.  (In principle `float(str(x))-x` should always be zero.  

```Python
>>> a = 99.0
>>> print(a)
99.00001
>>> print("%f" % a)
99.000006
>>> b = 99.00001
>>> print(b)
99.00001
>>> print(a - b)
-7.629395e-06
>>> import struct
>>> print(a, struct.pack("<f", a))
99.00001 b'\x00\x00\xc6B'
>>> print(b, struct.pack("<f", b))
99.00001 b'\x01\x00\xc6B'
```

dpgeorge, 2020-03-13:
I would guess this is a duplicate of #4212

goatchurchprime, 2020-03-13:
duplicate of #4212

closed 2020-03-13 by goatchurchprime