micropython/micropython #5758: Select and poll for UDP sockets always return HUP error

adamples, 2020-03-14:
Is seems that UDP sockets cannot be used in select/poll calls, because it will always return immediately, setting HUP error for the socket (which shouldn't ever happen for UDP).

For example:
```
>>> import usocket
>>> import uselect
>>> address = usocket.getaddrinfo('192.168.0.24', 4005)[0][-1]
>>> s = usocket.socket(usocket.AF_INET, usocket.SOCK_DGRAM)
>>> s.bind(address)
>>> uselect.select([s], [], [], 1000)
([], [], [<socket state=0 timeout=-1 incoming=0 off=0>])
```

(also note that s was included only in rlist)

Similarly with poll:
```
>>> poll = uselect.poll()
>>> poll.register(s, uselect.POLLIN)
>>> for event in poll.poll(): print(event)
... 
(<socket state=0 timeout=-1 incoming=0 off=0>, 16)
>>> 
```

The incorrect behaviour is observed for ESP8266 platform, using lwip stack:
```
>>> sys.platform
'esp8266'
>>> sys.implementation
(name='micropython', version=(1, 12, 0), mpy=9733)
```

The possible culprit is in modlwip.c: https://github.com/micropython/micropython/blob/master/extmod/modlwip.c#L1474
The code checks for socket state, but doesn't take into account that UDP sockets are always in `STATE_NEW`.

dpgeorge, 2020-03-16:
Thanks for the report.  I posted a fix for this in #5762.

adamples, 2020-03-16:
Great, thanks!

dpgeorge, 2020-03-17:
Fixed by 00267aae0b231cc4fec28ddaafd7f8917a75448f

closed 2020-03-17 by dpgeorge