micropython/micropython #5759: Non-blocking UDP socket returns ETIMEDOUT on recv instead on EAGAIN on ESP8266

adamples, 2020-03-14:
When reading from non-blocking UDP socket with no data available ETIMEDOUT error is returned. One would expect EAGAIN or EWOULDBLOCK in such situation.

For example:
```
>>> import usocket
>>> s = usocket.socket(usocket.AF_INET, usocket.SOCK_DGRAM)
>>> address = usocket.getaddrinfo('192.168.0.24', 4005)[0][-1]
>>> s.bind(address)
>>> s.settimeout(0)
>>> s
<socket state=0 timeout=0 incoming=0 off=0>
>>> s.recv(1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 110] ETIMEDOUT
```

The incorrect behaviour is observed for ESP8266 platform:
```
>>> sys.platform
'esp8266'
>>> sys.implementation
(name='micropython', version=(1, 12, 0), mpy=9733)
```

Looking at the modlwip.c code, there seems to be no special treatment applied for timeout value of 0:
https://github.com/micropython/micropython/blob/master/extmod/modlwip.c#L597

dpgeorge, 2020-03-17:
Thanks for the report.  A fix is posted in #5765.

dpgeorge, 2020-03-17:
Fixed by 8f0778b2092a4af1b7109b997a3c9c14766e31e8

closed 2020-03-17 by dpgeorge