micropython/micropython #5763: WLAN.connect causes the board to need to be reflashed

zurgeg, 2020-03-16:
Hello,
I have been attempting to create the network NET_AP_ESP32 by connected to my network and entering it's password, here is my understanding of it (And, no my wifi password is not password) :
```
import network

wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.scan()
wlan.connect('Wifi','password')

ap = network.WLAN(network.AP_IF)
ap.config(essid='NET_AP_ESP32')
ap.config(max_clients=10)
ap.active(True)
```
My network security is WPA2 Personal and my network name starts with a capital letter in case it helps.

But then it always disconnects from the wifi with code 15 then 205 and pressing ctrl+c doesn't terminate the code. ```ampy``` says it can't enter raw repl and I'm forced to reset (reflash) the board.

Thanks in advance!

zurgeg, 2020-03-16:
Ok, now it's not forcing me to reflash the board, but it still won't connect.

zurgeg, 2020-03-16:
Oh,  and I'm guessing some of you figured this out by my Wi-Fi hotspot name that my board is an ESP32 Thing from Sparkfun.

tve, 2020-03-16:
Dunno why you feel the need to reflash. Don't put the code into boot.py (or main.py) until it's stable, then at worst you need to press the reset button or use your tty pgm to toggle the line.

In terms of what you're trying to achieve, you need to set `ap.active(True)` before `ap.config()` and for me it doesn't take the `max_clients` param, but that may be due to my fork not having it. I don't know what the effect of `ap.active(True)` before configuring the AP is, doesn't make much sense to me, but experimentally it's what works. Someone would need to dig into ESP-IDF to see whether there's an alternate flow possible.

zurgeg, 2020-03-17:
@tve 

> Dunno why you feel the need to reflash

Using CTRL+C still gave me the error with couldn't enter raw repl

> Don't put the code into boot.py (or main.py) until it's stable

How can I tell if it's stable?

> you need to set ```ap.active(True)``` before ```ap.config()```

I did and my NET_AP_ESP32 still didn't show up

Also, it's a problem with the __station__ not the AP.

However, your suggestion is still good!

tve, 2020-03-17:
There was a bug with WPA/WPA2 mixed auth, are you using that or is your AP on WPA2-only? (It sounds like it, but not sure)

zurgeg, 2020-03-17:
Yes actually, my WI-Fi is actually a WPA/WPA2 WI-Fi.

tve, 2020-03-17:
You need to use a build with esp-idf v3 or the release version of esp-idf v4.

zurgeg, 2020-03-17:
I did use ESP-IDF3.

dpgeorge, 2020-03-20:
To work in both STA and AP mode at the same time it may be that the WiFi channel needs to be the same for both.  Check the channels that are being used.  Also, try running this code at the REPL without ampy and provide a full log of the debugging output, it should say why it disconnected.

zurgeg, 2020-03-20:
The reasons for disconnecting are 15 then 205:

zurgeg, 2020-03-25:
Also @dpgeorge, what do you mean by it needs to be the same Wi-Fi channel for both.

tve, 2020-03-25:
The Wifi radio can only operate on one channel, so STA and AP must use the same channel. When you operate in STA+AP mode the ESP-IDF automatically sets the AP's channel to the same as the STA's channel, the latter being dictated by the AP it's connecting to. There's nothing special for you to do.

PureBasic, 2020-05-19:
I have this kind of problem with an ESP32. ESP-IDF v4 Micropython v1.12... with spiram.
ESP hangs when calling wlan.connect('ssid','password').
For some reason, calling wlan.status() before calling wlan.connect() works.

tve, 2020-05-20:
can you paste the console output?

PureBasic, 2020-05-21:
Yeah I cannot reproduce the bug today. I dunno.
I'm really just starting using Micropython so I somehow may had have overlooked something or did something wrong yesterday...