micropython/micropython #5767: ESP32 BLE auto reconnection question

Walkline80, 2020-03-17:
Probably before the firmware v1.12-210, the hid device implemented by esp32 board can still automatically reconnect to my phone, of course, the board must be reset twice times. But after the firmware is upgraded to v1.12-256, every time this happens when "Auto Reconnect":

```docs
>>> activating ble...
>>> I (10070) BTDM_INIT: BT controller compile version [5aed448]
>>> I (20477) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
>>> I (20587) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
>>> GAP procedure initiated: stop advertising.
>>> ble activated
>>> services registed
>>> GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=7 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=160 adv_itvl_max=160
>>> advertising...
>>> [51:7C:9F:AB:EB:4C] connected, handle: 0
>>> [51:7C:9F:AB:EB:4C] disconnected, handle: 0
>>> GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=7 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=160 adv_itvl_max=160
advertising...
```

It looks like this, the board will still automatically reconnect, but it will disconnected immediately. In irq_central_disconnect callback, I coded to restart advertising, at this time, the phone can manually  connect the device again.

dpgeorge, 2020-03-17:
Can you provide full code to reproduce this issue?

Walkline80, 2020-03-18:
> Can you provide full code to reproduce this issue?

That's uh.....a little complex

1. Clone [this repo](https://gitee.com/walkline/ESP32-BLE-Remote_Controller.git)
2. Download [const.py](https://gitee.com/walkline/micropython-beacon-library/raw/master/ble/const.py) and [tools.py](https://gitee.com/walkline/micropython-beacon-library/raw/master/ble/tools.py), put both of them to repo's `ble` folder
3. Execute `ble/remote_controller.py`
4. After advertising, use mobile to scan and connect with `RemoteCon`
5. After connected(bonded), reset your development board(ESP32) and execute `remote_controller.py` again
6. After all.... you will see the output information just like I posted

Walkline80, 2020-03-18:
Additional problem: after many testing, include replace the keyboard, mouse, and consumer's `HID Descriptor`, only Android can receive data sent by the board, but not iOS devices

dpgeorge, 2020-03-18:
>  5 After connected(bonded), reset your development board(ESP32) and execute `remote_controller.py` again

What kind of "reset" do you do: a hard reset, power cycle, or soft reset?

> As owner of the phone, he will only found that the board will not reconnect to his phone, and he have to manually re-scan and re-connect by everytime the board reset, that's a sad story....

It sounds like that is expected behaviour because pairing/bonding is not implemented.

> after many testing, include replace the keyboard, mouse, and consumer's `HID Descriptor`, only Android can receive data sent by the board, but not iOS devices

Did this work with a previous esp32 firmware version?

Walkline80, 2020-03-18:
> 
> What kind of "reset" do you do: a hard reset, power cycle, or soft reset?
> 

A reset, just restart the board by pressing "EN" button on it, otherwise, re-plugging the USB is the same effect.

> 
> It sounds like that is expected behaviour because pairing/bonding is not implemented.
> 

No, because the board and phone have been properly connected and bonded once before

> 
> Did this work with a previous esp32 firmware version?
>

No, never successfully operated on an iOS device....only Android test successed

Walkline80, 2020-03-25:
@dpgeorge I have been tried the [esp-idf hid demo](https://github.com/espressif/esp-idf/tree/release/v3.3/examples/bluetooth/ble_hid_device_demo) on my ESP32 board, I have a question.

After scanning the device on iPhone and clicking the searched device to connect, a dialog box for requesting pairing will appear. You can choose to pair or ignore it, but the pairing request does not appear on the MicroPython firmware. Is this right? Is this related to some aspects of encryption?

martinlombana, 2021-03-08:
> It sounds like that is expected behaviour because pairing/bonding is not implemented.

Hi, @dpgeorge. Could you confirm if this is still the case for ESP32? It has not been implemented still? Or is there a nightly, more recent build where It would be implemented?

Thanks,

jimmo, 2021-03-11:
Yep, still the case for ESP32.

xyzzy42, 2023-11-06:
Pairing/bonding has been implemented for ESP32 since 5dbb822.

But it still doesn't seem to persist across resets.

While pairing works, and successfully does the key exchange and starts encrypting, there is no `_IRQ_SET_SECRET` event after pairing.  The only secret set is the local IRK when ble starts up.  Since the secret was not saved, of course the pairing can not be reconnected again after a reset and the secret has been lost.

On ESP32, Nimble is not part of the mpy source by part of ESP-IDF instead.  The ESP-IDF version has been modified.  One thing they did was create `ble_store_nvs.c` file that will store the ble secrets with the ESP NVS storage system.

I haven't traced it enough to see if the store callbacks are really not getting called by nimble and why that is, but I think it's probably due to this use of NVS.