micropython/micropython #5768: STM32F407VGT6 Support

montyx99, 2020-03-17:
Hello,

Sorry if my question is noob, but I'd like to learn to use micropython. I thought that to purchase one of the following boards:
https://www.aliexpress.com/item/4000094075769.html

There are 2 options to choose from:
1. STM32F407VGT6
2. STM32F407VET6

The F407VG has double size of Flash memory, so I prefer this one, if it's supported, but I've found the second type (STM32F407VET6) only in the documentation.
Please help me, which to buy.
Thanks a lot in advance!

mcauser, 2020-03-17:
Both boards work with MicroPython. For $2 more, I'd get the VG, with more flash.

montyx99, 2020-03-17:
Thanks a lot @mcauser. I'm going to buy the VG one.

closed 2020-03-17 by montyx99

mcauser, 2020-03-17:
Here's a board def for the VGT6: https://github.com/mcauser/MCUDEV_DEVEBOX_F407VGT6

This is for a v2.0 board, but it should be similar on the v3.0.
Double check the schematics and see if have moved around any pins.
Ask the seller if they can send you the schematics pdf. You can find the v2.0 one on my github repo.

Looking at the photos, they have removed the "DevEBox TM" from the top and bottom silk, so it might be a clone of the original MCUDev board. Some of the comments of people who received they board show photos with the DevEBox label, so uhh, I dunno. Wait and see what arrives...

montyx99, 2020-03-17:
I've found this:
[image]

montyx99, 2020-03-17:
If I may have an additional question, what does the 10/100Gb Ethernet with MAC support means? Is this possible to attach a simple RJ45 socket, and I can use the ethernet connection, or do I need an additional module?

mcauser, 2020-03-18:
I wouldn't trust photos which they could have copied from other vendors :)

I'm not sure about the Ethernet MAC support.
I don't think you can just expose 8 pins to a RJ45 to get Ethernet working.
You may have to use an external board, connected using RMII interface. eg.
http://blog.tkjelectronics.dk/2012/08/ethernet-on-stm32f4discovery-using-external-phy/

More about MAC and PHY.
The MAC is the Media Access Controller - Layer 2 in the OSI stack
and the PHY is the Physical Layer - Layer 1 in the OSI stack
https://e2e.ti.com/blogs_/b/analogwire/archive/2015/07/15/three-things-you-should-know-about-ethernet-phy
https://www.rfwireless-world.com/Terminology/PHY-vs-MAC.html

There were some pull requests which added Eth RMII support to a few boards:
https://github.com/micropython/micropython/pulls?q=RMII

montyx99, 2020-03-18:
Thank you very much for the very detailed answer!