micropython/micropython #5771: ESP32 taking pin back from adc to gpio requires machine.reset()

gmos, 2020-03-18:
When you use a Pin for ADC it can never be a gpio pin until a hard reset is done.
Example:

```
p = Pin(36, Pin.IN)
p()  # Yields 1 since it is conncted to 3.3V via external pull-up
a = ADC(Pin(36))
a.read()  # Yields 4095 on standard attenuation
del a
p = Pin(36, Pin.IN)
p()  # Yields 0 but should be 1
```
Cause: When the pin is taken for the ADC the primary mux is set to connect to the RTC_ADC.  Subsequent re-initialisation with Pin() or Pin.init does not switch the mux back.  This normally works because after reboot, by default  the primary mux connects to the GPIO mux.
As a solution we could reset the pin before selecting in `machine_pin_obj_init_helper()` in machine_pin.c.  This seems to work well:
```
    // configure the pin for gpio
    gpio_reset_pin(self->id);   # restores defaults, similar to reboot
    gpio_pad_select_gpio(self->id);
```

tve, 2020-03-18:
Looking at the MP docs for the `Pin` constructor:

> Access the pin peripheral (GPIO pin) associated with the given id. If additional arguments are given in the constructor then they are used to initialize the pin. Any settings that are not specified will remain in their previous state.

Note the last sentence. I'm curious how other ports solve this issue. Maybe adding a "reset=False" kw arg to the constructor and init documented as "resets the pin to power-up defaults before applying the specified configuration" is a solution? That would be another non-portable addition to the Pin class :-(.

gmos, 2020-03-18:
Good point Thorsten and to be solved easily.
The full reset of the pin was a bit overkill, and indeed would affect re-init of any pin.
By diving a bit deeper into the SDK, I found we can do:
```
#include "driver/gpio.h"
#include "driver/rtc_io.h"                             
...
    // configure the pin for gpio
    if (rtc_gpio_is_valid_gpio(self->id))   // Is it an RTC_ADC capable pin?
        rtc_gpio_deinit(self->id);          // Reset primary mux only. NOP if already GPIO pin.
    gpio_pad_select_gpio(self->id);
```
The init() semantics for GPIO pins is not changed now and the code it is even a wee bit more efficient.

dpgeorge, 2020-03-19:
Doing a `Pin(36, Pin.IN)` should reconfigure the GPIO fully so that it works in input mode (OTOH doing `Pin(36)` won't reconfigure anything).  Other ports (eg stm32) work this way and you can switch between digital, ADC, DAC, back to digital, etc.

So this is something that should be fixed for the esp32.

gmos, 2020-04-13:
@dpgeorge what do you want me to do?  Nothing or do a pull request?  If you want the latter, which solution would be best.  The first implements a total reset, as you state should be the case.  The second maintains upwards compatibility where existing programs do not expect a full reset.

tve, 2020-04-13:
> The second maintains upwards compatibility where existing programs do not expect a full reset.

I don't think this is an issue, the current implementation clearly has a bug that contradicts the docs and other implementations.

Given DP's clarification I do believe that a full pin reset is correct, but it should only happen if the mode is set in the call.

dpgeorge, 2020-04-16:
> I do believe that a full pin reset is correct, but it should only happen if the mode is set in the call.

Yes that's correct.  Feel free to open a PR, otherwise I'll eventually get to it.

jonathanhogg, 2020-11-09:
@dpgeorge I've just hit this problem myself, so I've submitted a PR with the suggested fix.

closed 2020-11-12 by dpgeorge

dpgeorge, 2021-04-15:
It looks like using `rtc_gpio_deinit()` was the right approach, see #7079 and #7121.

commit 8a917ad2 (2020-11-12): esp32/machine_pin: Reset pin if init sets mode.

This will forcibly grab the pin back from the ADC if it has previously been
associated with it.

Fixes #5771.