micropython/micropython #5778: lightsleep results in machine reset on ESP32

jdstroy, 2020-03-20:
Hello,

I'm running this version of MicroPython on ESP32:

`MicroPython v1.12-262-g19ea30bdd on 2020-03-20; ESP32 module with ESP32`

And I'm observing unexpected behavior with machine.lightsleep().

I would expect the following program code:

```
import machine
import time
while True:
    machine.lightsleep(10)
```

to behave in the same way as:

```
import machine
import time
while True:
    time.sleep_ms(10)
```

However, the prior always results in a WDT reset on my platform:

>>> import machine;
```
>>> while True:
...     machine.lightsleep(10);
...
...
...
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
```

This seems like a bug in machine.lightsleep(), as I can reproduce this with any value given to lightsleep, and even performing consecutive machine.lightsleep(100); calls directly without a loop.

jdstroy, 2020-03-20:
To be clear, I have not configured any WDTs from within Micropython.  I assume the log-mentioned RTCWDT is configured by FreeRTOS or Micropython outside of the Python environment.

dpgeorge, 2020-03-21:
> However, the prior always results in a WDT reset on my platform:

It's probably because the loop is starving the RTOS from doing work.  To test if this is the case try inserting a `time.sleep_ms(10)` after the ligthsleep.

jdstroy, 2020-03-22:
Agreed, it seems that the RTOS is being starved when machine.lightsleep() is called.

When time.sleep_ms(10) is added, the platform does still resets unexpectedly, but it takes a lot longer to get there.  WiFi gets dropped (as expected, since the wifi stack tasks won't be running), too.

```
>>> import machine
>>> import time
>>> while True:
...     machine.lightsleep(10);
...     time.sleep_ms(10);
...
...
...
I (135407) wifi: bcn_timout,ap_probe_send_start
I (137027) wifi: ap_probe_send over, resett wifi status to disassoc
I (137027) wifi: state: run -> init (c800)
I (137027) wifi: pm stop, total sleep time: 122771820 us / 137694919 us

I (137027) wifi: new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (137037) wifi: STA_DISCONNECTED, reason:200
beacon timeout
I (138357) wifi: STA_DISCONNECTED, reason:201
no AP found
I (138437) wifi: new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (138437) wifi: state: init -> auth (b0)
I (139087) wifi: state: auth -> init (200)
I (139087) wifi: new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (139087) wifi: STA_DISCONNECTED, reason:2
I (140407) wifi: STA_DISCONNECTED, reason:201
no AP found
I (141727) wifi: STA_DISCONNECTED, reason:201
no AP found
I (141807) wifi: new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (141807) wifi: state: init -> auth (b0)
I (142447) wifi: state: auth -> init (200)
I (142447) wifi: new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (142447) wifi: STA_DISCONNECTED, reason:2
I (143767) wifi: STA_DISCONNECTED, reason:201
no AP found
I (143847) wifi: new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (143847) wifi: state: init -> auth (b0)
I (144497) wifi: state: auth -> init (200)
I (144497) wifi: new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (144497) wifi: STA_DISCONNECTED, reason:2
I (145817) wifi: STA_DISCONNECTED, reason:201
no AP found
I (145897) wifi: new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (145897) wifi: state: init -> auth (b0)
I (146537) wifi: state: auth -> init (200)
I (146537) wifi: new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (146537) wifi: STA_DISCONNECTED, reason:2
I (147857) wifi: STA_DISCONNECTED, reason:201
no AP found
I (147937) wifi: new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (147937) wifi: state: init -> auth (b0)
I (148587) wifi: state: auth -> init (200)
I (148587) wifi: new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (148587) wifi: STA_DISCONNECTED, reason:2
I (149907) wifi: STA_DISCONNECTED, reason:201
no AP found
I (149987) wifi: new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (149987) wifi: state: init -> auth (b0)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5008
ho 0 tail 12 room 4
load:0x40078000,len:10600
ho 0 tail 12 room 4
load:0x40080400,len:5684
entry 0x400806bc
I (531) cpu_start: Pro cpu up.
I (531) cpu_start: Application information:
I (531) cpu_start: Compile time:     Mar 20 2020 00:39:39
I (535) cpu_start: ELF file SHA256:  0000000000000000...
I (540) cpu_start: ESP-IDF:          v3.3.1
I (545) cpu_start: Starting app cpu, entry point is 0x400836ec
I (0) cpu_start: App cpu up.

```

tve, 2020-03-22:
I believe you're driving the esp8266 into a regime it was not engineered for...
Maybe we can pop-up a level? What are you trying to accomplish? How are you trying to use light-sleep for what purpose? How do you expect to use Wifi with light-sleep?
I've written up a lot of what I know about the esp8266 and light-sleep in https://blog.voneicken.com/2018/lp-wifi-esp8266-assoc/ (this has nothing to do with MP). Maybe that helps inform what your goal is?

jdstroy, 2020-03-23:
Hello,

I'm not trying to accomplish anything with it right now... if I were, the solution seems pretty simple: replace calls to machine.lightsleep with time.sleep_ms.  I was exploring what's in MicroPython, and saw that lightsleep was new (I haven't updated in a long time).  The code snippets above are just me playing around and exploring, now reduced to a simple and reproducible example.

I'd be content with a nice warning in the docs that machine.lightsleep() may cause the platform to trigger a platform reset by starving the RTOS WDT, as it seems like this is unexpected behavior; but it'd be even nicer if that's taken care of for me.  Examples might be to adjust the WDT interval, or disarm the WDT during lightsleep (probably not a good idea), etc.