micropython/micropython #5779: machine.soft_reset() behaves curiously on esp8266

antariandel, 2020-03-20:
Hi!

This is related to https://github.com/micropython/micropython/issues/5764
Thanks a lot for implementing it by the way! 😄 

If I invoke it from the REPL, everything works as expected, but if it's invoked from a running code, it will drop to REPL (and with no **MPY: soft reboot** message). Here's an example:

    try:
        trixie.run()

    except KeyboardInterrupt:
        trixie.runner.stop()

    except trixie.ResetError:
        # land here on reset
        print('SOFT RESET')
        machine.soft_reset()

    except trixie.RebootError:
        # land here on reboot
        print('HARD RESET')
        machine.reset()

    except Exception as exc:
        # we should never end up here
        print('EXCEPTION')
        trixie.runner.stop()

trixie.run() raises trixie.SoftReset on soft reset intent, after which 'SOFT RESET' is printed, but then it drops to REPL instead.

I saw it in another issue, that it's by design that main.py won't be called on soft-reset, so I tested the snippet above in **_boot.py** and **boot.py** also, but it produces the same behavior. Not having 'MPY: soft reset' printed is suspicious. I'm confused if this is indeed a bug, or I'm doing something wrong.

Thanks a lot for looking into this!

dpgeorge, 2020-03-21:
> trixie.run() raises trixie.SoftReset on soft reset intent,

Can you provide the code that defines `trixie.SoftReset`, and also the code that raises it to trigger soft-reset intent?

antariandel, 2020-03-21:
Sure, but it'll be somewhat convoluted. 😄

Trixie is a project that I've been working on for the past month. It'll be a user code 'supervisor' with the ability to change user code over HTTP. The repository is here: https://github.com/antariandel/trixie-core (very much WIP).
The snippet above is in **src/boot.py**. The function **trixie.run()** is the main function, which starts it all. It sets up the network, the HTTP server and starts a bunch of coroutines to manage it all.
The relevant coro is **reboot()** inside **run()** from **trixie/\_\_init\_\_.py** , which raises **trixie.ResetError** with a delay.

That said, sorry, I made a mistake in my previous post, it's **trixie.ResetError**, not trixie.SoftReset, which is raised.

**reboot()** is a coroutine scheduled in uasyncio's main loop by **action_reboot()** (also in **run()**), so it should just propagate the error up to the main loop, stopping **loop.run_forever()** and escaping **run()**, which it must be doing, because I get 'SOFT RESET' printed as expected.
The error also gets propagated through a different chain. It gets passed to **runner.stop()**, which passes it to **code.cleanup()** (code.py is in **example/**), where it's discarded (in the current example).

**action_reboot()** is tied to a POST request to the URL **/sys/reboot/soft** (line 228 in trixie/\_\_init\_\_.py), so I trigger the whole thing with a request. Reboot (ie. **machine.reset()**) works by the way.

tve, 2020-03-22:
TBH I recommend you produce a small test case that enables others to repro and show the problematic output.

antariandel, 2020-03-22:
You're right, but I can't for the life of me reproduce it outside the above setup. I tried soft-resetting from function, from module and from frozen module.
Is there anything you suspect might be the culprit?