micropython/micropython #5780: wlan.ifconfig('dhcp') yields TypeError on esp8266

antariandel, 2020-03-20:
Hi!

I can't configure the interface back to DHCP, it yields:

    >>> wlan.ifconfig('dhcp')
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    TypeError: object 'str' isn't a tuple or list

wlan is WLAN(STA_IF)

dpgeorge, 2020-03-21:
> I can't configure the interface back to DHCP

This is currently not implemented on the esp8266, but it should be easy to implement it.

dpgeorge, 2020-03-21:
See #5781 for a proposed implementation of this feature.

antariandel, 2020-03-21:
Cool, thanks! I'll test it today or tomorrow.

antariandel, 2020-03-22:
Seems to work:

    >>> wlan.connect( [...] )
    >>>
    [...]
    dhcp client start...
    ip:192.168.1.100,mask:255.255.255.0,gw:192.168.1.1
    >>> wlan.ifconfig(('192.168.1.200', '255.255.255.0', '192.168.1.1', '192.168.1.1'))
    >>> wlan.ifconfig()
    ('192.168.1.200', '255.255.255.0', '192.168.1.1', '192.168.1.1')
    >>> wlan.ifconfig('dhcp')
    >>> wlan.ifconfig()
    ('192.168.1.100', '255.255.255.0', '192.168.1.1', '192.168.1.1')
    >>>

dpgeorge, 2020-03-24:
Thanks for testing!  Fixed by 9fa32169e9e6cd23710a2a68c05aad96da22aae5

closed 2020-03-24 by dpgeorge