micropython/micropython #5783: [Errno 103] ECONNABORTED\r\n') after some time

gary094, 2020-03-22:
I can't get behind the following issue:

I run this script:

```import machine
import network
from time import sleep
from urllib.urequest import urlopen
import json

wifiap = network.WLAN(network.AP_IF)
wifiap.active(False)
routercon = network.WLAN(network.STA_IF)
routercon.active(True)
routercon.ifconfig(('10.0.0.128','255.255.255.0','10.0.0.138','10.0.0.138'))
routercon.connect('mywifi', '123')
while not routercon.isconnected():
    pass

posturl=('http://10.0.0.156:23102/rest/v2/send')

adc = machine.ADC(0)
gc.enable()

while True:
    value = adc.read()
    if value < 200:
       message = {'username': 'test', 'message': value, 'chatid': 'test', 'password': 'test', 'notifyself': 'false'}
       r = urlopen(posturl, data=json.dumps(message).encode())
       r.close()
    gc.collect()
    sleep(1)
```

And after some period of time it errors with ECONNABORTED.
It usually happens when the "urlopen" line has not been executed for a while.

When this happens I can't do http posts to this endpoint anymore 
`r = urlopen(posturl, data=json.dumps(message).encode())`

however if I use the domain pointing to that IP instead it starts to work again:

`r = urlopen("http://webservice.ddns.net:23102/rest/v2/send", data=json.dumps(message).encode())`

How can I solve this issue permanently without reseting the board each time?

tve, 2020-03-22:
which platform are you using?

gary094, 2020-03-22:
ESP8266 nodemcu with Micropython "esp8266-20191220-v1.12.bin"

DavePlayz, 2020-03-25:
Ohh, I understand how this can be solved, just lemme check if it works.

DavePlayz, 2020-03-25:
Nope, my solution broke.

gary094, 2020-03-25:
Alright, what was your proposed solution? Were you able to reproduce the issue?

dpgeorge, 2020-03-27:
This is most likely an issue with intermittent internet connectivity.  To write robust code assume that at any point the WiFi may go down and exceptions may be raised in code that accesses the network.

gary094, 2020-03-27:
While I agree with you to raise exceptions (I already did improve the code lately) the wireless connection stays active and working all the time. To prove that:

I tried to spam my webservice for 10 minutes straight with requests each second doing this:
`r = urlopen("http://10.0.0.156:23102/rest/v2/send", data=json.dumps(message).encode())`

which works totally fine. Afterwards if I let it sit idle for a some time and try again it stops working and giving me the ECONNABORTED error.

Now, interestingly when this happens I can still **use the domain** pointing to that **same** webservice:
`r = urlopen("http://webservice.ddns.net:23102/rest/v2/send", data=json.dumps(message).encode())`

I also tried to reproduce this issue with other clients, like a raspberry pi 3B to no avail.

I can't explain how it can't reach something on my local network (10.0.0.156) but still at the same time 100% of the time being able to reach the domain pointing to the exact same webservice

tve, 2020-03-27:
> I can't explain how it can't reach something on my local network (10.0.0.156) but still at the same time 100% of the time being able to reach the domain pointing to the exact same webservice

What do you mean by "at the same time"? Do you have test code that does that? Can you post it? If you're typing into the repl, it's definitely not "at the same time"... Maybe you can paste a transcript? For example, when it's in the failed state can you type in the two forms of urlopen and show what happens? If you do that, also check that `socket.getaddrinfo('webservice.ddns.net', 23102)[0][-1]` returns the expected IP address.

gary094, 2020-03-27:
I tested manually as well as using test code.

Let me explain:
Imagine running a endless loop like this:

```
while True:
     sleep(1)
     try:
         r = urlopen("http://10.0.0.156:23102/rest/v2/send", data=json.dumps(message).encode())
     except Exception as e:
         print(str(e))
     finally:
         r.close()
```

I could spam this webservice probably all day long. At least I tested it for 15 minutes straight without any signs of exhaustion.

I break the loop (CTRL-C) and wait for a couple of minutes (usually 5-10 minutes)

Now wanting to issue a http post again I do this:

```
>>> r = urlopen("http://10.0.0.156:23102/rest/v2/send", data=json.dumps(message).encode())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "urllib/urequest.py", line 28, in urlopen
OSError: [Errno 103] ECONNABORTED
>>> 
```

which returns: ` [Errno 103] ECONNABORTED`

However using the public facing domain pointing to this webservice:

```
>>> r = urlopen("http://webservice.ddns.net:23102/rest/v2/send", data=json.dumps(message).encode())
>>> 
```

is **always** able to connect.

>  when it's in the failed state can you type in the two forms of urlopen and show what happens? 

failed state locally:

```
>>> r = urlopen("http://10.0.0.156:23102/rest/v2/send", data=json.dumps(message).encode())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "urllib/urequest.py", line 28, in urlopen
OSError: [Errno 103] ECONNABORTED
>>> 
```

failed state using public facing domain ~2 seconds after the above:

```
>>> r = urlopen("http://webservice.ddns.net:23102/rest/v2/send", data=json.dumps(message).encode())
>>>
```

> `socket.getaddrinfo('webservice.ddns.net', 23102)[0][-1]`

Yes this returns the IP mapped to the domain "webservice.ddns.net" (some router in between forwarding port 23102 to 10.0.0.156)

Finally I can prove that when the esp8266 is in failed state still other clients are able to issue a http post, e.g. a raspberry pi 3B

esp:

```
>>> import os
>>> os.uname()
(sysname='esp8266', nodename='esp8266', release='2.2.0-dev(9422289)', version='v1.12 on 2019-12-20', machine='ESP module with ESP8266')
>>> r = urlopen("http://10.0.0.156:23102/rest/v2/send", data=json.dumps(message).encode())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "urllib/urequest.py", line 28, in urlopen
OSError: [Errno 103] ECONNABORTED
>>> 
```

raspberry:
```
>>> import os
>>> os.uname()
posix.uname_result(sysname='Linux', nodename='raspberryPi', release='4.19.66-v7+', version='#1253 SMP Thu Aug 15 11:49:46 BST 2019', machine='armv7l')
>>> r = urlopen("http://10.0.0.156:23102/rest/v2/send", data=json.dumps(message).encode())
>>>
```

So I can rule out the webservice being flaky.

tve, 2020-03-27:
When you show
```
>>> r = urlopen("http://webservice.ddns.net:23102/rest/v2/send", data=json.dumps(message).encode())
>>>
```
is the request actually being performed, e.g., can you see it in your server logs or is it perhaps just encountering a silent error in the DNS lookup and not telling you making it look like it succeeded?

gary094, 2020-03-27:
Yes the request is being performed. While doing these tests I monitor the logs and verify that the request found its destination.

gary094, 2020-04-02:
Alright so I tried to find out more via tcpdump by monitoring the responsible eth interface:
this is the working state:

```
10:41:46.251226 IP (tos 0x0, ttl 254, id 47, offset 0, flags [none], proto TCP (6), length 44)
    10.0.0.128.3623 > 172.18.0.3.8080: Flags [S], seq 6877, win 2144, options [mss 536], length 0
10:41:46.251282 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto TCP (6), length 44)
    172.18.0.3.8080 > 10.0.0.128.3623: Flags [S.], seq 3928427302, ack 6878, win 64240, options [mss 1460], length 0
10:41:46.253524 IP (tos 0x0, ttl 254, id 48, offset 0, flags [none], proto TCP (6), length 40)
    10.0.0.128.3623 > 172.18.0.3.8080: Flags [.], seq 6878, ack 3928427303, win 2144, length 0
10:41:46.560092 IP (tos 0x0, ttl 254, id 49, offset 0, flags [none], proto TCP (6), length 225)
    10.0.0.128.3623 > 172.18.0.3.8080: Flags [P.], seq 6878:7063, ack 3928427303, win 2144, length 185: HTTP, length: 185
	POST /rest/v2/send HTTP/1.0
	Host: 10.0.0.156
	Content-Length: 124
	
	{"chatid": "test", "password": "test", "notifyself": "false", "username": "test", "message": 114}[!http]
10:41:46.560154 IP (tos 0x0, ttl 64, id 54622, offset 0, flags [DF], proto TCP (6), length 40)
    172.18.0.3.8080 > 10.0.0.128.3623: Flags [.], seq 3928427303, ack 7063, win 64055, length 0
10:41:46.575772 IP (tos 0x0, ttl 64, id 54623, offset 0, flags [DF], proto TCP (6), length 132)
    172.18.0.3.8080 > 10.0.0.128.3623: Flags [P.], seq 3928427303:3928427395, ack 7063, win 64055, length 92: HTTP, length: 92
	HTTP/1.1 200 
	Content-Length: 0
	Date: Thu, 02 Apr 2020 08:41:46 GMT
	Connection: close
	
10:41:46.576162 IP (tos 0x0, ttl 64, id 54624, offset 0, flags [DF], proto TCP (6), length 40)
    172.18.0.3.8080 > 10.0.0.128.3623: Flags [F.], seq 3928427395, ack 7063, win 64055, length 0
10:41:46.584405 IP (tos 0x0, ttl 254, id 50, offset 0, flags [none], proto TCP (6), length 40)
    10.0.0.128.3623 > 172.18.0.3.8080: Flags [F.], seq 7063, ack 3928427395, win 2052, length 0
10:41:46.584440 IP (tos 0x0, ttl 64, id 54625, offset 0, flags [DF], proto TCP (6), length 40)
    172.18.0.3.8080 > 10.0.0.128.3623: Flags [.], seq 3928427396, ack 7064, win 64055, length 0
10:41:46.584942 IP (tos 0x0, ttl 254, id 51, offset 0, flags [none], proto TCP (6), length 40)
    10.0.0.128.3623 > 172.18.0.3.8080: Flags [.], seq 7064, ack 3928427396, win 2051, length 0
10:41:51.496083 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has 172.18.0.3 tell 172.18.0.1, length 28
10:41:51.496120 ARP, Ethernet (len 6), IPv4 (len 4), Reply 172.18.0.3 is-at ff:ff:ff:ff:ff:ff, length 28
```

and this is what tcpdump filters when in broken state:

```
$ sudo tcpdump -nnvvSK -i br-0e95fa4cc63f not host 172.18.0.2 and not host 172.18.0.4
[sudo] password for :         
tcpdump: listening on br-0e95fa4cc63f, link-type EN10MB (Ethernet), capture size 262144 bytes

10:11:50.933336 IP (tos 0x0, ttl 254, id 39, offset 0, flags [none], proto TCP (6), length 44)
    10.0.0.128.5579 > 172.18.0.3.8080: Flags [S], seq 6715, win 2144, options [mss 536], length 0
10:11:50.933403 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto TCP (6), length 44)
    172.18.0.3.8080 > 10.0.0.128.5579: Flags [S.], seq 2522914169, ack 6716, win 64240, options [mss 1460], length 0
10:11:50.985835 IP (tos 0x0, ttl 254, id 630, offset 0, flags [none], proto TCP (6), length 40)
    10.0.0.128.5579 > 172.18.0.3.8080: Flags [R.], seq 6716, ack 2522914170, win 2144, length 0
10:11:52.049716 IP (tos 0x0, ttl 254, id 40, offset 0, flags [none], proto TCP (6), length 44)
    10.0.0.128.5579 > 172.18.0.3.8080: Flags [S], seq 6715, win 2144, options [mss 536], length 0
10:11:52.049779 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto TCP (6), length 44)
    172.18.0.3.8080 > 10.0.0.128.5579: Flags [S.], seq 2540357592, ack 6716, win 64240, options [mss 1460], length 0
10:11:52.099267 IP (tos 0x0, ttl 254, id 631, offset 0, flags [none], proto TCP (6), length 40)
    10.0.0.128.5579 > 172.18.0.3.8080: Flags [R.], seq 6716, ack 2540357593, win 2144, length 0
10:11:53.197735 IP (tos 0x0, ttl 254, id 41, offset 0, flags [none], proto TCP (6), length 44)
    10.0.0.128.5579 > 172.18.0.3.8080: Flags [S], seq 6715, win 2144, options [mss 536], length 0
10:11:53.197800 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto TCP (6), length 44)
    172.18.0.3.8080 > 10.0.0.128.5579: Flags [S.], seq 2558295385, ack 6716, win 64240, options [mss 1460], length 0
10:11:53.223505 IP (tos 0x0, ttl 254, id 632, offset 0, flags [none], proto TCP (6), length 40)
    10.0.0.128.5579 > 172.18.0.3.8080: Flags [R.], seq 6716, ack 2558295386, win 2144, length 0
10:11:54.424572 IP (tos 0x0, ttl 254, id 42, offset 0, flags [none], proto TCP (6), length 44)
    10.0.0.128.5579 > 172.18.0.3.8080: Flags [S], seq 6715, win 2144, options [mss 536], length 0
10:11:54.424640 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto TCP (6), length 44)
    172.18.0.3.8080 > 10.0.0.128.5579: Flags [S.], seq 2577464756, ack 6716, win 64240, options [mss 1460], length 0
10:11:55.432101 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto TCP (6), length 44)
    172.18.0.3.8080 > 10.0.0.128.5579: Flags [S.], seq 2577464756, ack 6716, win 64240, options [mss 1460], length 0
10:11:55.648710 IP (tos 0x0, ttl 254, id 43, offset 0, flags [none], proto TCP (6), length 44)
[... 1 lines cut ...]
10:11:55.648773 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto TCP (6), length 44)
[... cut ...]
    10.0.0.128.5579 > 172.18.0.3.8080: Flags [S], seq 6715, win 2144, options [mss 536], length 0
10:11:58.024262 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto TCP (6), length 44)
    172.18.0.3.8080 > 10.0.0.128.5579: Flags [S.], seq 2633708854, ack 6716, win 64240, options [mss 1460], length 0
10:11:58.054911 IP (tos 0x0, ttl 254, id 637, offset 0, flags [none], proto TCP (6), length 40)
    10.0.0.128.5579 > 172.18.0.3.8080: Flags [R.], seq 6716, ack 2633708855, win 2144, length 0
```

I can't spot any noticable problems by this output, do you?

tve, 2020-04-03:
Interesting... Look at the 3rd packet, the reset sent by the esp8266, it's ACK field reflects the initial seq picked by the server in the second packet (esp_ack=srv_seq+1), so it correctly received the syn-ack (2nd packet). The question then is why does it reset the connection instead of proceeding?

If you ask me, I would put my money on running out of memory. I just had that experience on the esp32 where it couldn't allocate a network buffer and proceeded to shut down the wifi STA interface and then bring it back up. Symptom at application level was ECONNABORTED, just like here. I would bet something similar is happening here (I don't think it drops the wifi sta because the next connection attempt happens a second after the reset, which is too soon to reconnect unless perhaps you are on an open wifi network). Note that I'm talking about the system needing to allocate a network buffer, not python, so it's the system memory heap, not the python one (I don't know how they're managed on the esp8266).

gary094, 2020-08-08:
Alright I'm coming back to this issue once again.
I appreciate your input however its hard for me to believe this issue is caused by memory problems.

So I made a class where I do something like this:

```
class Class:

    def send(self, message, chatid):
        payload = {
            'username': self.username,
            'password': self.password,
            'message': message,
            'chatid': chatid
        }

        try:
            r = urequests.post('http://webservice.ddns.net:23102/rest/v2/send', data=json.dumps(payload).encode(), headers={'Content-Type': 'application/json'})
            print(payload)
        except Exception as e:
            print(e)
```

Then in `main.py` I initialize this class like this:

`myobject = Class('name', 'password'`

and use the `send()` function like this:

`myobject.send('name', 'message')`

and this returns 100% of the time:
`[Errno 103] ECONNABORTED`

**however**

defining this exact function `send()` **without using class** directly in `main.py` gives me better results and it does not return that error above.

Could it still be memory issues then?

peterhinch, 2020-08-09:
Maintaining a communications link over WiFi is not easy as brief outages can occur especially near the limit of range.

With @kevinkk525 I developed two solutions capable of maintaining a link through WiFi outages and other disruptions characteristic of radio links. These have been tested over long periods on ESP8266, ESP32 and Pyboard D. See [resilient MQTT](https://github.com/peterhinch/micropython-mqtt) and [micropython-iot](https://github.com/peterhinch/micropython-iot).

In testing the only platform exhibiting rock-solid reliability over weeks of continuous communication was the Pyboard D. ESP32 occasionally reboots and ESP8266 sometimes crashes. Typically these events occur once every few days.

gary094, 2020-08-09:
Thank you but I'm pretty sure there is no problems with my wireless network as I'm able to communicate with my service at all times (Other devices, from the public facing domain or from the internal network, ...). I am starting to think my ESP8266 board has bugs. See my answer above.

I'm ordering a ESP32 now, hoping that these issues go away.

In summary: ESP8266 can't for some reason execute a `urequests.post` wrapped in a function inside a class but for some reason it absolutely can do that without using a class. I would appreciate if somebody can give me an explanation about why this is happening, otherwise I write it off as a hardware failure

peterhinch, 2020-08-10:
There is a big difference between "hardware" devices like ESP8266, ESP32, and Pyboard D when compared with "computer" devices like laptops, tablets and phones. Hardware devices experience brief outages not seen by computers, presumably because the OS hides them from the user until their duration exceeds some threshold.

My WiFi as seen by computers is rock solid. In recent testing of ESP32, brief outages occurred when the device was in the same room as the AP. To maintain a continuous link there is no choice other than to code round outages. Hence our libraries.

shr00mie, 2020-09-10:
the following may have in part or whole been pulled entirely out of my ass:

i've been running into the same issue across a whole bunch of projects. LED lights with HA, a remote pulsating beacon connected to AdafruitIO, an automated herb watering project...symptoms are quite similar to those previously mentioned:

- project runs anywhere from 20-30 hours, at which point it throws ECONNABORTED and just freezes.

per previous recommendations, the following code was implemented post module import and within the various loops:

```python
# in main.py after primary module import
gc.collect()
	gc.threshold(gc.mem_free() // 4 + gc.mem_alloc())

# somewhere in main loop:
gc.collect()
```

when this still did not solve the issue, i decided to start sending back gc.mem_free() and gc.mem_alloc() and throw them in influx. here's the output from the last session:

[image]

the 12-E was turned on at 13:07 on the 8th and crapped out at 9:56 on the 9th, so we're looking at ~21 hours.

per the free & allocated memory, it does appear that `gc.collect()` is indeed doing its job, but as previous posts have hinted at, the devil is probably in the details, and at least if i'm understanding things correctly, while X amount of free space may be available, if the required amount of contiguous space is not available, then you're SOL.

at least based on my...cursory experimentation and the symptoms, i would tend to agree with the out of memory hypothesis, but not as it relates to total available, but contiguous available depending on how much the process is trying to write. would be helpful to know how much that network process is trying to write perhaps? Anyone have that on hand or might be able to investigate? At least at that point maybe something can be...i dunno...preallocated so there's always X amount of contiguous memory reserved for that?

next up i'll see about trying to log contiguous memory during operation in a similar manner to see if that will yield and valuable input.

if any of the above is just wrong or based on faulty assumptions, by all means please educate me.

VineetS086, 2021-01-22:
I'm facing the exact error with esp8266....

`
from urequests import get

get('http://192.168.29.157:8888/2/?format=json').json()`

this code sometimes returns this error "OSError: [Errno 103] ECONNABORTED"
but mostly it takes too long to execute...(could be few ms or even tens of seconds in some cases)

My wifi is fine...
should I use esp32 instead of esp8266 or the problem can be resolved by some other method...?

peterhinch, 2021-01-22:
Re the "out of memory" hypothesis. On an ESP8266 29K of free RAM is usually plenty. You're right in stating that memory failures can occur due to a lack of contiguous free RAM, but I doubt this is happening here. If a MicroPython allocation fails, you get an explicit "allocation failed" error message.

VineetS086, 2021-01-22:
> Re the "out of memory" hypothesis. On an ESP8266 29K of free RAM is usually plenty. You're right in stating that memory failures can occur due to a lack of contiguous free RAM, but I doubt this is [...]

@peterhinch I do believe that ram is plenty, what's the problem then...?

The same script was working fine yesterday...(Before I ran It in while loop with 0.4 sec sleep delay)
Is it possible that I may have damaged the board...?😅
I'm using my laptop to power it... with 22 Ω resistors in series with  LEDs...

When The problem continues I reset the board and it is resolved...

This is my main.py... 

`pn_1 = Pin(5, Pin.OUT)
pn_2 = Pin(4, Pin.OUT)
pn_3 = Pin(0, Pin.OUT)
pn_4 = Pin(14, Pin.OUT)
pn_5 = Pin(12, Pin.OUT)
pn_6 = Pin(13, Pin.OUT)
pn_7 = Pin(15, Pin.OUT)
pn_8 = Pin(10, Pin.OUT)
sleep(1)
do_connect()

def loop():
    while 1:
        try:
            sleep(0.6)
            respon = get('http://192.168.29.157:8888/2/?format=json').json()['appliance_set']
            status = [i['status']  for i in respon]
            
            pn_1.value(status[0])
            pn_2.value(status[1])
            pn_3.value(status[2])
            pn_4.value(status[3])
            pn_5.value(status[4])
            pn_6.value(status[5])
            pn_7.value(status[6])
            pn_8.value(status[7])
        
        except:
            sleep(2)
            do_connect()
            
            
loop()

`

gary094, 2021-01-22:
> should I use esp32 instead of esp8266 or the problem can be resolved by some other method...?

I was not able to resolve this issue on the esp8266 boards.
However after testing this exact same script on multiple esp32 boards I can confirm it does occasionally happen, **but** a `gc.collect()` is fixing the issue and freeing some memory. The esp32 boards unfortunately have other quirks...

Of course I tried this way before on the esp8266 boards as well however it does not actually do anything it seems. No memory gets freed as far as I can tell.

VineetS086, 2021-01-22:
hmmm. Turns out it may have been the ram...
After I improved my code, this issue stopped(at least for now)...

I used gc.collect()
and used some tips from [Damien George's Talk](https://www.youtube.com/watch?v=hHec4qL00x0)...

ohidurbappy, 2022-03-30:
I ran into the same issue on my LAN.

Then I tried ngrok tunnel and it works just fine.

evgeniyzha, 2023-07-20:
I have almost exactly the same issue, but in a slightly different setup. I am running a http server on the esp8266 and when I try to respond, if the operation on the esp takes a bit longer (usually north of 10sec), the client socket is sometimes closed (though at the time it is created upon client request, the instance prints `timeout=-1`; I suspect it's an issue with the underlying RTOS on the esp, which closes "inactive" sockets once in a while especially when some memory threshold is reached).

This is a hack that made things much better for me:
- first, in my loop, which looks very similar to gary094's loop (I am also reading the adc repeatedly), I introduced `gc.collect()` for each 20 iterations using the modulo operator (I also tired other values, but 20 seemed a good tradeoff between not doing it every loop and still improving the OSError frequency)
- and then, during debugging of this error, I actually printed the available memory, also each 20 iterations right after the `gc.collect()` and strangely enough this yielded by far the best results

so putting this in my loop:
```
i = (i+1) % 20
if not(i):
    gc.collect()
    print("free memory is {} Bytes".format(gc.mem_free()))
```
improved things significantly. NOTE: I am not using the REPL output in any way, so the printing is going 'nowhere'; to avoid the MCU cycles needed for that I also tried assigning the `gc.mem_free()` value to some variable and then e.g. returning it, so it does not get optimised away by the mpy compiler as I suspected this call is the main part of the "solution" - but not printing it gave a much higher failure rate (`OSError` popped up much more often). Just to give you a sense of the stats:
- if my loop takes 20s on the esp without the hack above, the `OSError` appears almost each request
- with `gc.collect()` each **20** loop cycles, the `OSError` appears in about 4 out of 10 requests; almost same number of failures if `print(...)` is omitted 
- with `gc.collect()` **and** `print("free memory is {} Bytes".format(gc.mem_free()))` each **100** loop cycles, the `OSError` appears in about 3 out of 100 requests
- with `gc.collect()` **and** `print("free memory is {} Bytes".format(gc.mem_free()))` each **20** loop cycles, the `OSError` appears in about 3 out of 1000 requests

Not a solution, but an OKish "workaround" for me until this bug is fixed and maybe useful for anyone stumbling upon the same issue and/or for chasing down the root cause.

evgeniyzha, 2023-07-20:
BTW, the "RTOS does it" hypothesis would help to explain...

- the different behaviour of RTOS boards vs bare metal
> In testing the only platform exhibiting rock-solid reliability over weeks of continuous communication was the Pyboard D. ESP32 occasionally reboots and ESP8266 sometimes crashes. Typically these events occur once every few days.

> I was not able to resolve this issue on the esp8266 boards.
> However after testing this exact same script on multiple esp32 boards I can confirm it does occasionally happen

- the "soft link" between memory / `gc`
> per previous recommendations, the following code was implemented post module import and within the various loops:

> **but** a `gc.collect()` is fixing the issue and freeing some memory.

- as well as the connection to sleep = inactivity
> I've also implemented `sleep_type(SLEEP_NONE)` 

- and also why encapsulated post does not work (as the memory overhead of the class might trigger the RTOS "socket purging")
> In summary: ESP8266 can't for some reason execute a `urequests.post` wrapped in a function inside a class but for some reason it absolutely can do that without using a class

Unfortunately, I am not enough of an RTOS or micropython implementation expert to help pinpoint it in the code, but as micropython does not just silently fail but is able to raise an `OSError`, I assume there must be a way of monitoring the socket status from the micropython implementation code

Also, if indeed the RTOS does it, it does so "silently" as seen here:
> Alright so I tried to find out more via tcpdump by monitoring the responsible eth interface:

and also supported by the behaviour of my system: I use the python `requests` module for interacting with the esp from a RPi with a timeout set higher than the operation runs on the esp board; whenever the esp fails with an `OSError`, the request on the RPi is still "stalled" and times out - so the RPi does not see any termination TCP package (no "FIN" headers)

jonnor, 2025-03-03:
How to reproduce this issue?