micropython/micropython #5787: ESP32 RMT Produces unexpected result when loop is True

peterhinch, 2020-03-23:
See [this forum post](https://forum.micropython.org/viewtopic.php?f=18&t=8010)
```python
import esp32
from machine import Pin

r = esp32.RMT(0, pin=Pin(27), clock_div=80)
r.loop(1)
r.write_pulses((100, 100, 10, 10), start=0) 
```
The waveform shows alternate cycles of 100/100μs one correctly followed by a 10/10μs pulse, the other missing this pulse.

Tested on the reference board, firmware MicroPython v1.12-211-g9c07c973c on 2020-03-03; ESP32 module with ESP32.

peterhinch, 2020-03-25:
[This](http://hinch.me.uk/ESP32/iss5787.png) is what I'm seeing, confirmed on a scope. Could this be related to https://github.com/micropython/micropython/issues/5623?

DavePlayz, 2020-03-25:
Possibly, if the pulse is generated like that, I think it will break, whereas I can't think of a solution to this...

carstenblt, 2020-03-26:
Maybe related to https://github.com/espressif/esp-idf/issues/4664?

dpgeorge, 2020-06-19:
Should be fixed by 3a9d948032e27f690e1fb09084c36bd47b1a75a0

closed 2020-06-19 by dpgeorge