micropython/micropython #5790: Esp32 using MQTT help

mohammedabujarad, 2020-03-24:
hello everyone,

i have an Esp32 that is connected to a dht22 sensor which publish the data to an mqtt server, also i have another esp32 that subscribe to the topic where the data are published, but the esp32 that subscribe to the topic it disply an error says ( mqtt: OSError(-1,) ). i am using micro python for the esp32 and both of them are connected to a raspberry pi 3 as server. also both of the board have two script. the first one connect too the WIFI, and for the second script. the first one publish the data and the second one subscribe to the topic.

here are the codes:

the code that publish the data:

```
from time import sleep
from umqtt.robust import MQTTClient
from machine import Pin
from dht import DHT22

SERVER = '172.20.10.11' # MQTT Server Address 
CLIENT_ID = 'ESP32_DHT22_Sensor'
TOPIC = b'temp_humidity'

client = MQTTClient(CLIENT_ID, SERVER)
client.connect() # Connect to MQTT broker

sensor = DHT22(Pin(15, Pin.IN, Pin.PULL_UP)) # DHT-22 on GPIO 15 (input with internal pull-up resistor)

while True:
try:
sensor.measure() # Poll sensor
t = sensor.temperature()
h = sensor.humidity()
if isinstance(t, float) and isinstance(h, float): # Confirm sensor results are numeric
msg = (b'{0:3.1f},{1:3.1f}'.format(t, h))
client.publish(TOPIC, msg) # Publish sensor data to MQTT topic
print(msg)
else:
print('Invalid sensor readings.')
except OSError:
print('Failed to read sensor.')
sleep(2)
```
the code that subscribe to the topic:

```
import time
from umqtt.robust import MQTTClient

SERVER = '172.20.10.11' # MQTT Server Address 
CLIENT_ID = 'ESP32_DHT22_Sensor'
TOPIC = b'temp_humidity'

def sub_cb(topic, msg):
print((topic, msg))

c = MQTTClient(CLIENT_ID, SERVER)

c.DEBUG = True
c.set_callback(sub_cb)

if not c.connect(clean_session=False):
print("New session being set up")
c.subscribe(TOPIC)

while 1:
c.wait_msg()

c.disconnect()

```
here is the error that i get whenever i ran the second code:

mqtt: OSError(-1,)
I (27817) wifi: ampdu: ignore deleting tx BA0
mqtt: OSError(-1,)
mqtt: OSError(-1,)
mqtt: OSError(-1,)
I (41297) wifi: ampdu: ignore deleting tx BA0
mqtt: OSError(-1,)
mqtt: OSError(-1,)
mqtt: OSError(-1,)
I (55127) wifi: ampdu: ignore deleting tx BA0
mqtt: OSError(-1,)
mqtt: OSError(-1,)
mqtt: OSError(-1,)
mqtt reconnect: OSError(23,)
I (68337) wifi: ampdu: ignore deleting tx BA0
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)
mqtt reconnect: OSError(23,)

tve, 2020-03-24:
I think your question is much more suited to the forum https://forum.micropython.org than here as it's not a bug per-se. If you browse the forum a bit you'll see many similar questions ("umqtt.robust is not robust") and the answer invariably is to use mqtt_as with asyncio...

mohammedabujarad, 2020-03-25:
Thank you for the response, I will ask my question there.

tve, 2020-03-25:
Forums thread: https://forum.micropython.org/viewtopic.php?f=2&t=8028
Issue can be closed, IMHO

closed 2020-03-25 by mohammedabujarad