micropython/micropython #5794: Is there a way to use MicroPython like a flash drive?

zurgeg, 2020-03-25:
So, I have a Sparkfun ESP32 Thing, and am trying to use it as if it were a flash drive. Let me explain what I mean. So, basically I want a drive named MY_ESP_DRIVE or something. Then I put file ```a.py``` on it then I import a.py in my code. Is this possible. I know circuit python has this functionality built in. If this is not possible, could you add it?
Thanks in advance.

MrSurly, 2020-03-25:
This is more of a hardware question.  CircuitPython _boards_ have MCUs with dedicated USB hardware that can emulate a flash drive, and custom firmware to support that.  The Sparkfun board uses an ESP32 (with no USB capability), and an external USB-to-serial IC to allow connectivity to  a host -- it can only ever show up as a serial port to the host.

This isn't something that will be added to mainline MicroPython anytime soon, if ever, though I'll defer to @dpgeorge.

Having said that, you can do something similar (with the Sparkfun device) with [mpy-utils](https://github.com/nickzoic/mpy-utils).

dhalbert, 2020-03-25:
MicroPython PyBoards and various other MCU boards with native USB has this capability already. A PyBoard shows up as PYBFLASH, for instance.

Note that both the host computer and the MicroPython program can write to the drive, and unless it's changed recently, there's no locking, so the FAT filesystem can get corruptedif both are writing simultaneously. It's hard to do the locking automatically because the interface is block-oriented, and it's not easily possible to determine when a filesystem transaction is done.

In CircuitPython by default we make the host have read/write access and the program has read-only access. This can be flipped in `boot.py` so the host is read-only and Python has r/w, but then you have to have a way to undo this to edit your programs again.

zurgeg, 2020-03-26:
@MrSurly I would like to note that on the ESP32’s page (I don’t know which one or where) it says that there is a way to use the board’s flash like a thumb drive. I think maybe we could implement something like that to achieve this.

jimmo, 2020-03-26:
@zurgeg if you can find a link then that would help. My guess would be that they're referring to the ESP32-S2 (which isn't available yet) which has been announced to have this functionality.

Just to re-iterate the points from above - the ESP32 does not have a USB peripheral, so it is not possible for the ESP32 to do this. The way that ESP32 boards support a USB UART is always via an external chip, such as an FTDI or CP210x.

CircuitPython on SAMD, NRF (and possibly others, but not ESP32 which is not to my knowledge supported by CircuitPython), and MicroPython on STM32 support the thumb drive functionality. I understand that CircuitPython is planning to support the ESP32-S2.

zurgeg, 2020-03-28:
For those of you wondering, I found a (potential) answer that wasn’t the one I saw before but it’s at https://hackaday.io/project/58788-esp8266-with-true-usb. Also what you could do as an alternative is use MicroPython’s serial functionality (I don’t know if it exists though, however I am pretty sure it does) to make an Arduino send a message with the file contents to the board, as I know a lot of Arduinos support Native USB. This may not be suitable for larger files, such as images but it works until MicroPython supports the Native USB functions of the ESP32-S2. However, I am still leaving this open as someone may come up with a superior solution.

zurgeg, 2020-04-08:
Also, I might purchase a SAMD/NRF board; How might I go about doing this on one of those?

jimmo, 2020-04-08:
USB MSC is only supported (on MicroPython) on STM32. Pyboard being the obvious easy way to get started (as they come with MicroPython pre-installed), but there are all sorts of different STM32 development boards. Boards based on the F4, F7, and some L4 are your best options.

For SAMD, CircuitPython will do this on many SAMD21 and SAMD51 boards. https://circuitpython.org/downloads   I'm not sure about which nRF boards they support (if so it would only be the ones with USB peripherals, like the nRF52840). The MicroPython SAMD and nRF do not (yet) support MSC.

FYI this might be better moved into the forum -- https://forum.micropython.org/

closed 2020-04-08 by jimmo

dhalbert, 2020-04-08:
> For SAMD, CircuitPython will do this on many SAMD21 and SAMD51 boards. https://circuitpython.org/downloads I'm not sure about which nRF boards they support (if so it would only be the ones with USB peripherals, like the nRF52840). The MicroPython SAMD and nRF do not (yet) support MSC.

CircuitPython only supports boards that can do MSC. We dropped support for nRF52832 abd ESP8266 for that reason. We support a bunch of STM32F4 boards, and are working on STM32F7 and H7. Also we support Sony Spresense, and we're starting on i.MX 10xx chips, with support for boards like the Teensy 4.0. There's also an initial port to the LiteX FPGA-based RISC-V board. We're planning to support ESP32-S2, the new version of ESP32 that has USB support.