micropython/micropython #5797: extmod/uasyncio: wait_for cancellation bugs

kevinkk525, 2020-03-25:
CancelledError forwarding if wait_for got cancelled (not the task it awaits):

Your assumption "# Ignore CancelledError from aw, it's probably due to timeout" in funcy.py line 20 https://github.com/micropython/micropython/blob/ad004db662b383a9bf310601e13e896c99e11e30/extmod/uasyncio/funcs.py#L20 is incorrect.
As you can see in the testcase "wait_for.test_cancellation_forwarded", if you cancel the caller task of "wait_for", the coro awaited by "wait_for" gets correctly cancelled, but the caller of "wait_for" never gets the Exception raised. The CancelledError gets swallowed by "wait_for" instead of forwarding it. 

Wait_for also behaves incorrectly if CancelldError Exceptions are caught inside the task it awaits because they don't get raised to wait_for anymore. However Cpython does also raise those to wait_for even if they are caught in the task wait_for awaits.

All my wait_for testcases (working on upy and cpy) are here: https://gist.github.com/kevinkk525/3313a8d2763031a705e37d1748d1d996

The one in question for all bugs mentioned:
```
async def test_cancellation_forwarded(catch=False, catch_inside=False):
    res = [False, False]

    async def wait():
        nonlocal res
        try:
            await asyncio.wait_for(awaiting(2, catch_inside), 0.5)
        except asyncio.TimeoutError:
            dprint("Got timeout error")
            res[0] = asyncio.TimeoutError
            raise
        except asyncio.CancelledError:
            dprint("Got canceled")
            if catch:
                res[0] = True
            else:
                raise

    async def killer(t):
        dprint("killer started")
        await asyncio.sleep(0.2)
        dprint("killing wait()")
        t.cancel()

    t = asyncio.create_task(wait())
    k = asyncio.create_task(killer(t))
    try:
        await t
    except asyncio.CancelledError:
        dprint("waiting got cancelled")
        res[1] = asyncio.CancelledError
    return res
```

dpgeorge, 2020-04-12:
Thanks for the report.  I have been looking in to this and the hard thing here is not the forwarding of cancellation but rather the order in which cancellation propagates to various parts.

In CPython the `wait_for` does not wait on the input awaitable, so if the `wait_for` is cancelled then it gets the cancellation first, and forwards it to the awaitable, which gets it second.

But in MicroPython's current implementation of uasyncio, `wait_for` waits on the awaitable so if `wait_for` is cancelled then the awaitable gets the cancellation first.

Fixing that is tricky (it requires making an intermediate `Event` which is set either from the timeout or the awaitable finishing, and `wait_for` waits on this `Event`).

If the order of cancellation doesn't matter (ie is an implementation detail) then the fix is simpler.  I'm not sure if the order is critical for this case...

dpgeorge, 2020-04-13:
See related CPython bug here: https://bugs.python.org/issue32751

kevinkk525, 2020-04-13:
This is indeed interesting but actually only part of the problems. 

This is the current output of the test-script on upy:
```
729388027331 Test: test_cancellation_forwarded-catch
729388028371 killer started
729388028921 awaiting started
729388229106 killing wait()
729388229578 awaiting canceled
729388230422 Test 'test_cancellation_forwarded-catch' result: False, Got: NameError('local variable referenced before assignment',), expected: [True, False]

729388233102 Test: test_cancellation_forwarded
729388233682 killer started
729388234194 awaiting started
729388435165 killing wait()
729388435625 awaiting canceled
729388436380 Test 'test_cancellation_forwarded' result: False, Got: NameError('local variable referenced before assignment',), expected: [False, <class 'CancelledError'>]

729388438752 Test: test_cancellation_forwarded-catch_catch-inside
729388439381 killer started
729388439818 awaiting started
729388640301 killing wait()
729388640687 awaiting canceled
729388641367 Test 'test_cancellation_forwarded-catch_catch-inside' result: False, Got: [False, False], expected: [True, False]

729388642672 Test: test_cancellation_forwarded_catch-inside
729388643273 killer started
729388643708 awaiting started
729388843584 killing wait()
729388843968 awaiting canceled
729388844640 Test 'test_cancellation_forwarded_catch-inside' result: False, Got: [False, False], expected: [False, <class 'CancelledError'>]
```

The first tests do not even run correctly because there is a bug in wait_for due to incorrect assumption in: https://github.com/micropython/micropython/blob/ad004db662b383a9bf310601e13e896c99e11e30/extmod/uasyncio/funcs.py#L20
This gets triggered when wait_for itself gets cancelled. It then just ignores the cancellation and returns "res" which is not defined. Instead, wait_for needs to distinguish between itself being cancelled and the CancelledError from the coro it awaits due to the killer killing it.
I made a PR to fix this: #5918

The other problems that arise from the awaited task catching the cancelledError and not forwarding it can be explained by the different behaviour in CPython (what you posted). There's probably no easy fix for that and it is a rare scenario. It is generelly unwise to catch CancelledErrors without raising them as it could result in unintended behaviour anyway. So this could just stay there with a general warning about not catching CancelledErrors without raising them again.

commit b5059710 (2020-12-02): extmod/uasyncio: Fix cancellation handling of wait_for.

This commit switches the roles of the helper task from a cancellation task
to a runner task, to get the correct semantics for cancellation of
wait_for.

Some uasyncio tests are now disabled for the native emitter due to issues
with native code generation of generators and yield-from.

Fixes #5797.
closed 2020-12-02 by dpgeorge