micropython/micropython #5802: code-format.sh doesn't fix spaces for C-style casts

stinos, 2020-03-26:
```
        float r =  (  float  )    0.0;
        float r =   (float )    0.0;
        float r =   ( float)    0.0;
```
gets formatted into
```
        float r = (  float  )    0.0;
        float r = (float )    0.0;
        float r = ( float)    0.0;
```

stinos, 2020-03-26:
Here's another one:
```
(( mod < zero ) != (*y < zero ))
```
doesn't remove the space before mod and after zero.
So it's not the casting, it's the braces.

dpgeorge, 2020-03-26:
See #5804 for a fix.  Didn't try the above examples but it should hopefully address them all.

stinos, 2020-03-26:
That fixes the last one.
With these 2
```
# Add or remove space after C/D cast, i.e. 'cast(int)a' vs. 'cast(int) a' or
# '(int)a' vs. '(int) a'.
sp_after_cast                   = remove   # ignore/add/remove/force

# Add or remove spaces inside cast parentheses.
sp_inside_paren_cast            = remove   # ignore/add/remove/force
```
the casts are fixed for me as well (assuming we do not want a space after a cast)

dpgeorge, 2020-03-27:
> With these 2 ... the casts are fixed for me as well (assuming we do not want a space after a cast)

Ok, added those extra options to #5804

PR #5804 (2020-03-28): all: Remove spaces inside parenthesis.
Using a new option enabled in the uncrustify configuration.  Should fix #5802.
closed 2020-03-28 by dpgeorge