micropython/micropython #5808: ussl.wrap_socket fails with mbedtls_ssl_handshake errors on ESP32

czAdamV, 2020-03-26:
In some cases, the `ussl.wrap_socket` function fails with a nondescript error:
```
>>> import ussl, usocket
>>> s = usocket.socket()
>>> s.connect(('185.8.236.214', 443))
>>> s = ussl.wrap_socket(s, server_hostname="aladinonline.androworks.org")
mbedtls_ssl_handshake error: -4290
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 5] EIO
```
This doesn't seem to be a network problem as requests to other pages work well in the same session. This is in fact the only page I've run into this issue with, and I can't figure out what's wrong with its TLS setup.

I initially ran into this when using the `urllib.urequest.urlopen` function and I believed this was a micropython-lib bug, so I didn't post the issue here (original issue was micropython/micropython-lib#374), but I found out later that the bug wasn't present in the Micropython version 1.12 and that it was probably introduced by commit 3032ae115, which switches the version of ESP-IDF used to build Micropython for ESP32. Even though I managed to replicate the bug only using the undocumented parameter `server_hostname` of `wrap_socket` function, I really think this is a bug in Micropython (or some underlying library), not in micropython-lib.

tve, 2020-03-27:
I have good news and bad news for you ;-)
The good news is that I'm working on a PR to fix the obscure mbedtls error numbers. Here's what my version (using ESP_IDFv4) says:
```
MicroPython v1.12-307-gad004db66-dirty on 2020-03-26; ESP32 module with ESP32
Type "help()" for more information.
>>> import ussl, usocket
>>> s = usocket.socket()
>>> s.connect(('185.8.236.214', 443))
>>> s = ussl.wrap_socket(s, server_hostname="aladinonline.androworks.org")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: RSA - The publicration failed : BIGNUM - Memory allocation failed
```
The bad news, other than the funny typo, is that the error is due to too small a buffer. MP uses the asymmetric buffer sizes and that puts a limit on the RSA key size, or something like that (someone please correct me). This is a well-known issue with TLS and constrained memory devices. I believe the buffer sizes are fixed at compile time (that's a mbedtls limitation).
Hope this helps.

Carglglz, 2020-03-27:
Hi @czAdamV you may be want to read this thread #5543

@tve 

> The good news is that I'm working on a PR to fix the obscure mbedtls error numbers

This is great! Thanks for taking the time! 🥇 👍

enesbcs, 2021-05-14:
Maybe related to https://github.com/micropython/micropython/issues/7038

jimmo, 2022-07-21:
This issue is now being tracked in https://github.com/micropython/micropython/issues/8940.

closed 2022-07-21 by jimmo