micropython/micropython #5814: New uasyncio frozen in firmware has broken NUCLEO_L073RZ build.

andrewleech, 2020-03-27:
The tiny stm32 NUCLEO_L073RZ board is pretty close to the line at the best of times.

https://github.com/micropython/micropython/commit/35e2dd0979f80df8cb5fbb22a9766ee13d4ef437 has pushed it over again

https://gitlab.com/alelec/micropython_ci/-/jobs/485320119
```
 CC /builds/alelec/micropython_ci/micropython_stm32_NUCLEO_L073RZ_ad004db6/pins_NUCLEO_L073RZ.c
 LINK /builds/alelec/micropython_ci/micropython_stm32_NUCLEO_L073RZ_ad004db6/firmware.elf
 arm-none-eabi-ld: /builds/alelec/micropython_ci/micropython_stm32_NUCLEO_L073RZ_ad004db6/firmware.elf section `.text' will not fit in region `FLASH'
 arm-none-eabi-ld: region `FLASH' overflowed by 116 bytes

```

dpgeorge, 2020-03-28:
I did test this build locally, but my compiler is probably a different version that produces smaller code.

dpgeorge, 2020-03-28:
See #5823 for a fix.

andrewleech, 2020-03-28:
Ah good point, I should try updating the compiler version, I've got a newer Docker image ready to go already.

dpgeorge, 2020-03-28:
Fixed by 83461e37d4b27142faab2b1d91bd1462a66107e1

closed 2020-03-28 by dpgeorge