micropython/micropython #5821: how to escape run_forever() in extmod/uasyncio

antariandel, 2020-03-27:
Hi!

I hope you're not thinking I'm spamming. Tell me to stop if you think I shouldn't submit issues regarding `extmod/uasyncio` yet. In that case I'll go back to using `micropython-lib/uasyncio`.

So raising an error in a task does not stop the loop as it does in `micropython-lib/uasyncio`. What is the proper way to stop and exit from `loop.run_forever()`?

Thanks,
Zsolt

kevinkk525, 2020-03-27:
You are right to ask, in CPython you would do it with loop.stop() but it is currently not implemented: https://github.com/micropython/micropython/blob/9418611c8aa860812e28b6e2b6bfde7be5817b43/extmod/uasyncio/core.py#L223

dpgeorge, 2020-03-30:
See #5842 for an implementation of `stop()`.

dpgeorge, 2020-04-01:
`Loop.stop()` was implemented in b389bc0afaaa5a16dc11daaed056d89ea3da3b83

closed 2020-04-01 by dpgeorge

antariandel, 2020-04-03:
Cool! Thanks!