micropython/micropython #5824: thread_stacksize1 is still occasionally failing

dlech, 2020-03-28:
Even after #5784, the `thread_stacksize1` test is still failing sometimes.

Example https://travis-ci.org/github/micropython/micropython/jobs/667965401#L1801

dpgeorge, 2020-03-28:
Looks like it's on the standard, 64-bit unix build.  If it's not reproducible then it's hard to fix...

dpgeorge, 2020-03-28:
I can reproduce a crash that looks very similar using FreeBSD 12.0 on a 64-bit virtual machine.  There the stack size defaults to 64k.  Using 4k instead of 2k stack size in the test gets it passing, but I'd like to understand why.

dpgeorge, 2020-03-28:
Looking into the crash under FreeBSD, it occurs in `free(th)` in `mp_thread_finish()`.  It looks like the threads allocated stack is indeed too small and executing the thread overwrites data on the C stack.

So it seems that 8192 bytes are needed for the thread stack-overflow safety margin.

dpgeorge, 2020-03-28:
See #5826 for a proposed fix.

PR #5826 (2020-03-30): unix/mpthreadport: Ensure enough thread stack to detect overflow.
To fix #5824.
closed 2020-03-30 by dpgeorge