micropython/micropython #5827: something goes wrong when i want to run micropython on my own stm32 board

CCHLENG, 2020-03-28:
Hi !
For some reason, I can't use pyboard in my application. I need to make my own stm32 board to run micropython.
Now I'm trying to download the firmware to my board. The board I use has a STM32F407VGT6, same as [STM32F4DISC](https://www.st.com/en/evaluation-tools/stm32f4discovery.html).
So I try to download `STM32F4DISC-20200328-v1.12-310-g9418611c8.dfu` to the board with `DfuSeDemo` :
- set BOOT0 = 1, BOOT1 = 0
- power on
- download the dfu file with DfuSeDemo
- power off after download
- set BOOT0 = 0, BOOT1 = 0
but nothing happens when I power on. i did't see a new USB flash drive on my PC.

[image]

So what can i do? is there a guide to follow if i want to run micropython on my own board?

CCHLENG, 2020-03-28:
same way works on F407ZGT6.. so strange

dpgeorge, 2020-03-28:
This kind of problem is best suited to discuss at https://forum.micropython.org

Here is for bug reports and feature requests.

closed 2020-03-28 by dpgeorge