micropython/micropython #5828: extmod/uasyncio: Queue is missing

tve, 2020-03-28:
The new uasyncio doesn't have a Queue class (unless I'm getting blind?). That seems like a pretty core sync primitive. Should this be added or should that go into a separate library? There's also Condition, Semaphone and more... Where should such a library go or what is the plan?

kevinkk525, 2020-03-28:
I'd prefer if those primitives could be added to the firmware the same way lock and event are added.
Peter Hinch already has an implementation of those primitives, it just needs adapting to the new structure:
https://github.com/peterhinch/micropython-samples/tree/master/uasyncio_iostream/uasyncio

peterhinch, 2020-03-29:
It does appear to be there, along with Event and Lock. Alas the other CPython synchronisation primitives (Semaphore, BoundedSemaphore, Condition) are not.
```python
>>> for x in sorted(dir(uasyncio)):
...     print(x)
... 
CancelledError
Event
IOQueue
Lock
Loop
Queue
Server
SingletonGenerator
Stream
StreamReader
StreamWriter
Task
TimeoutError
__class__
__file__
__name__
__path__
core
create_task
event
gather
get_event_loop
lock
open_connection
ph_delete
ph_meld
ph_pairing
run
run_until_complete
select
sleep
sleep_ms
start_server
stream_awrite
sys
ticks
ticks_add
ticks_diff
wait_for
>>> 
```

tve, 2020-03-29:
Hmm, we must have different firmware versions... Or you have it in the filesystem and it gets pulled in. Can you locate the source code for me?
```
>>> import uasyncio
>>> for x in sorted(dir(uasyncio)): print(x)
...
CancelledError
Event
IOQueue
Lock
Loop
SingletonGenerator
Task
TaskQueue
TimeoutError
__class__
__file__
...
```

kevinkk525, 2020-03-29:
@peterhinch you have a lot of entries I can't find either. 
The Queue, StreamReader, StreamWriter, stream_awrite, ...

peterhinch, 2020-03-30:
Oops. Apologies you guys are right - I had an old version in my filesystem.

The loss of StreamReader and StreamWriter will break a lot of code. They exist in CPython.

kevinkk525, 2020-03-30:
> The loss of StreamReader and StreamWriter will break a lot of code. They exist in CPython.

If they exist in CPython, then it would be good if they get added to uPy too, especially because they were available before.