micropython/micropython #5831: Trailing .0 in float literal can cause rounding error

dlech, 2020-03-28:
This is more of a problem now that we are using `black` to format code since it doesn't allow floating point numbers with only a trailing `.` it always adds a `0` after the `.`.

The error comes from `mp_parse_num_decimal()` where the number is multiplied by 10 for each trailing 0 before being divided down back to the original number later.

https://github.com/micropython/micropython/blob/f6375ac3ebac28656a0a757952d32c265b1ba7aa/py/parsenum.c#L254

Would it be too much of a performance hit to check that all digits after `.` are 0? Or perhaps just test for the case where there is only one digit after `.` and that it is `0`?

_Originally posted by @dlech in https://github.com/micropython/micropython/issues/4212#issuecomment-605515703_

dpgeorge, 2021-05-18:
> Would it be too much of a performance hit to check that all digits after `.` are 0? Or perhaps just test for the case where there is only one digit after `.` and that it is `0`?

No I don't think so.  As per https://github.com/micropython/micropython/pull/5995#issuecomment-624732274 I think this will fix some but break others (float parsing).  See #6024 for a proper fix.

closed 2021-05-18 by dpgeorge

dpgeorge, 2022-08-12:
Fixed by 6cd2e4191803e95580bdfc57c06ea818454a25d1