micropython/micropython #5835: extmod/modussl_mbedtls: change getpeercert for a callback

tve, 2020-03-29:
The mbedtls module provides an undocumented `getpeercert` method. I'd like to remove it and replace it with a callback passed into wrap_socket. Since it's undocumented, there should be no harm, right? :-) :-)

Why would I want to do this? Well, https://tls.mbed.org/kb/how-to/reduce-mbedtls-memory-and-storage-footprint says:
> By default, Mbed TLS saves a copy of the peer certificate for the lifetime of an SSL session and makes it available through the public API mbedtls_ssl_get_peer_cert(). If the application does not need to inspect the peer certificate, disabling the compile-time option MBEDTLS_SSL_KEEP_PEER_CERTIFICATE saves RAM as the SSL module will not keep a copy of the peer certificate after the handshake. The API mbedtls_ssl_get_peer_cert() to obtain the peer certificate is still present, but always returns NULL.

I just took a random sample: the peer cert for micropython.org is 1363 bytes long. I only created the socket, connected, and wrapped it and at some point there were only 4.7KB free in the esp-idf heaps... So those 1363 bytes do count IMHO.

Why do I want to add a callback? That doc further says:
> If you need to inspect the peer certificate during or immediately after the handshake, you may still disable MBEDTLS_SSL_KEEP_PEER_CERTIFICATE and inspect the certificate through a verification callback instead. A verification callback is a function called during the verification of the peer certificate chain and can be registered via mbedtls_ssl_conf_verify().

I'm happy to submit a PR for this, thoughts on how to navigate the deprecation of the `getpeercert` method?

tve, 2020-03-29:
I came up with another idea:
- compile with MBEDTLS_SSL_KEEP_PEER_CERTIFICATE disabled
- use a callback to store the cert in ussl
- change getpeercert to return the cert and remove the reference in ussl to the cert, i.e., it will return the cert only once
- add getpeercert to docs and explain the read once semantics

The net effect is that current code continues to work unless it calls getpeercert repeatedly, which is unlikely. Code that wishes to save some memory can call getpeercert and drop/delete the return value.

felixdoerre, 2023-08-04:
Just for reference: `MBEDTLS_SSL_KEEP_PEER_CERTIFICATE` was always enabled by default, just the mbedtls version was too old to recognize it. On current main, mbedtls has been upgraded, so `MBEDTLS_SSL_KEEP_PEER_CERTIFICATE` has been set automatically, and `getpeercert` is broken and just returns none on current master. So this issue has been "solved" by accident.

dpgeorge, 2023-12-18:
> `MBEDTLS_SSL_KEEP_PEER_CERTIFICATE` was always enabled by default, just the mbedtls version was too old to recognize it. On current main, mbedtls has been upgraded, so [...]

Ah, thanks for the explanation!  I didn't realise this config option changed in this way.

Actually, in ef996d15b9cbadee591a185f27fb16e90a5d4f5d I made it so the `.getpeercert()` method is only included if `MBEDTLS_SSL_KEEP_PEER_CERTIFICATE` is enabled.  So that method is now removed in most builds (I think only esp32 enables this option).

PR #12259 (2024-02-07): extmod/modssl_mbedtls: Add cert verification callback.
the update of mbedtls has activated MBEDTLS_SSL_KEEP_PEER_CERTIFICATE. This results in the old getpeercert function not working anymore. On top, storing the certificate for the whole duration of the connection is wasting RAM.

This PR enables access to the certificate through a callback so it does not have to be stored if it is not needed, an can be inspected and analyzed in python while the connection is established, optionally aborting the connection.

fixes #5835
closed 2024-02-07 by dpgeorge