micropython/micropython #5843: extmod/uasyncio: Loop exits if only Tasks awaiting Events are added

kevinkk525, 2020-03-30:
If only Tasks are added to the Loop that are waiting for an Event to be set (or a Lock to be acquired but that makes no sense at all), the loop exits because those Tasks get removed from the main queue and put onto the waiting queue of the Event.
Use-case: Event is waiting to be set from an ISR. Will probably only ever be used without other Tasks running in minimal test-cases but would then be a problematic bug.

Expected behaviour: Loop continues to run. This needs #5795 (Events from ISR). Otherwise this problem will of course never occur.

```
>>> import uasyncio as asyncio
>>> ev=asyncio.Event()
>>> async def wait():
...     print("started waiting")
...     await ev.wait()
...     print("got event")
...
...
...
>>> asyncio.run(wait())
started waiting
>>>
```

agners, 2020-05-03:
Encountered this issue when I tried using `Event` for Bluetooth.

> This needs #5795 (Events from ISR). Otherwise this problem will of course never occur.
From what I understand if `micropython.schedule()` is used in a ISR, and then Event.set() called from there, this would be safe with todays `Event` implementation wouldn't it?

From what I understand the Bluetooth ISR (my case at hand) is actually called in schedule context and should be safe today.

miketeachman, 2021-02-18:
It would be great to see this fixed at some point.  It's a tough one to debug when you don't know about it.