micropython/micropython #5847: extmod/uasyncio Please reinstate StreamReader and StreamWriter

peterhinch, 2020-03-31:
The removal of these breaks existing code. It also breaks CPython asyncio compatibility.

andrewleech, 2020-04-01:
In my use case, I replace StreamReader and StreamWriter usages with `from uasyncio.stream import Stream` and they seem to work fine.

Is this compatible enough that it can be used as-is for compatibility, eg
`uasyncio/stream.py`
``` python
######
# Legacy uasyncio compatibility

Stream.aclose = Stream.wait_closed
Stream.awrite = stream_awrite
Stream.awritestr = stream_awrite  # TODO explicitly convert to bytes?

StreamWriter = Stream
StreamReader = Stream
```

and then have these as objects which can be imported on-demand in `uasyncio/__init__.py`

tve, 2020-04-01:
Can you just do something like:
```
>>> from uasyncio.stream import Stream as StreamReader
>>> from uasyncio.stream import Stream as StreamWriter
```

kevinkk525, 2020-04-01:
> Can you just do something like:

It's not about finding workarounds. Those objects extisted with the old uasyncio and exist in CPython so it should work the same with the new uasyncio version( without using workarounds that are basically breaking compatibility)

dpgeorge, 2020-04-01:
Sorry, these were actually there in a prior version but were removed (by me) in error.  So they should be put back, and can simply be aliases of `Stream`.

(Side note: at some stage during CPython's recent development `StreamReader` and `StreamWriter` were merged/replaced with a single `Stream` but that never made it into a CPython release.)

dpgeorge, 2020-04-01:
Fixed by f97b5395ed4b03d8cce6fda0c4b41d8118843690

closed 2020-04-01 by dpgeorge