micropython/micropython #5854: touchPad on Pin 2 (ESP32)

GitHubForTester, 2020-04-01:
I have downloaded the actual version of Micropython for the ESP32 and i wanted to use Pin 2 as a touchPin. I can decelerate it as a touchpin like this 'touchPad = TouchPad(Pin(2))'

But I cannot read from that: 'touchPad.read()'

Than I get an error: 'ValueError: Touch pad error'

Have anyone an idea?

tve, 2020-04-01:
Please use forums.micropython.org for this type of question. Thanks!

GitHubForTester, 2020-04-02:
I think it's an issue, because pin 2 is a touchpin, and its works, with other Pins its works,

HarshitDaftary, 2020-10-03:
@GitHubForTester , Did you get solution ? I also face the same error.

codemee, 2021-03-21:
I think it's the onboard LED connected to gpio02 cause the problem.

codemee, 2021-03-21:
You can use other touch pad instead of gpio02.

codemee, 2021-03-21:
Or use another type of board without LED connected to gpio02.