micropython/micropython #5857: extmod/uasyncio Provide means of scheduling I/O with high priority

peterhinch, 2020-04-02:
This was discussed at length in the past, and prototyped with early code of this uasyncio. There is known demand for it in high speed UART applications and audio processing. Salient points:

- Enable I/O readiness to be tested on every pass of the scheduler.
- The facility to be available on a per-interface basis. A subset of active interfaces can be selected to run at high priority.

This would improve real time throughput and reduce buffering requirements

hardkrash, 2020-04-03:
- Be communication interface agnostic
    - Should allow for any interface where an external action triggers communications. 
    - e.g. UART RX, CAN, I2C Slave, I2S, etc...

What about bluetooth data received?

What about ADC measurements and the like?  I'm thinking about free running ADC measurements where an interrupt regularly occurs but we a stream of data.

tve, 2020-04-03:
Let me play devils' advocate: are scheduling priorities the best solution? Specifically, would you prefer scheduling priorities or would you prefer soft interrupts (with atomic sections)? How about task priorities?

- Without any of these the max interrupt latency is the sum of all the tasks' longest non-blocking section.
- With scheduling priorities it becomes the longest non-blocking section across all tasks.
- With soft interrupts the latency becomes just the longest atomic section.

Thoughts about this:
- Only soft interrupts provide any practical upper bound on interrupt latency IMHO. Putting an upper bound on the longest non-blocking section of code is quite difficult and can easily change  unexpectedly when any piece of code in the app is changed.
- In the new asyncio the avg number of tasks in the run queue should be significantly lower than with the old one because there should no longer be a need for tasks to poll on I/O. So I would expect that scheduling priorities make less of a difference.
- Unless I'm missing something, fast_io does not implement task priorities, it really implements wake-up priorities. I.e., when a task blocks on something the wake-up action can specify the priority of the run queue to use. This means that if the task blocks on anything else it looses its priority. This is different from having task priorities where a setting on the task determines which run queue it gets placed into when it is awakened.
- Task priorities lead to interesting further problems such as prioritiy inversions and a variety of possible scheduling algorithms...

I don't find the trade-offs that easy to evaluate. One thing that might help is to have a couple of realistic application examples to think through?

miketeachman, 2020-04-04:
> One thing that might help is to have a couple of realistic application examples to think through?

I use fast_io in an ESP32 Application for **continuous and gapless processing of I2S audio data from a MEMS microphone**.  Audio samples are streamed into GPIO of the ESP32 and are buffered into DMA memory.  The DMA memory must be serviced at a fast rate to avoid over-running the DMA buffer (e.g.  losing audio samples).  In an application with several low-priority co-routines there is a risk that the DMA servicing coro is preempted by multiple low-priority coros... leading to an over-run of the DMA buffer.  The situation gets worse as the sampling rate of the microphone increases.  One of my micropython projects uses the fast_io implementation to create two high-priority coros, for UART and I2S handling.  https://github.com/miketeachman/micropython-street-sense/blob/master/streetsense.py.  Using fast_io allows the application to achieve continuous and gapless audio sample processing, while at the same time reading other sensors, updating a display, and using MQTT to push results to a database using WIFI.

It would be great to see this capability integrated into the new uasyncio work.

tve, 2020-04-04:
Thanks for the example and link. Looks like a cool app!
I tried to understand the code, am I getting this right:
- I2S runs at 10k samples/sec, 2x32-bits per sample (L+R)?
- 64 dma buffers of 256 samples each, 128KB total?
- the DMA buffers hold just under 1.7 seconds of data?
- the Microphone.run_mic coro sits in a loop, calls I2C.readinto() in order to read from the DMA buffers into a 2KB buffer and if there is no data it sleeps for 2ms using fast_io and tries again?
- if you got data, you write it to an sdcard, which is a totally blocking operation which can take over 100ms?
- so fast_io only comes into play in that when you sleep for 2ms it wakes up on the io priority queue?
- given that you have 1.7 seconds of buffer, is the 2ms sleep really critical?
- in the end, it seems you really wanted a high priority task, there is no interrupt latency in play?

peterhinch, 2020-04-04:
> Unless I'm missing something, fast_io does not implement task priorities, it really implements wake-up priorities.

It prioritises I/O streams only. I/O has a realtime aspect: the aim is to reduce latency relative to normal **uasyncio** scheduling. The reduction is of course poorer than an ISR. The benefit relative to an ISR is in avoiding the concurrency and allocation difficulties associated with code running in an interrupt context. While a soft ISR would overcome the allocation problem there would still be thread-safety problems which don't exist in the **fast_io** approach.

> Putting an upper bound on the longest non-blocking section of code is quite difficult and can easily change unexpectedly when any piece of code in the app is changed.

Nobody said realtime programming was easy :)

miketeachman, 2020-04-04:
@tve  great analysis of the code.  It works as you indicated.
> * given that you have 1.7 seconds of buffer, is the 2ms sleep really critical?

It's not critical for a 10 kHz sample rate, where the 2k samples are accumulated in ~25ms.  It starts to matter for faster sampling rates.  For example, with a 44.1 kHz sample rate, 2k samples are accumulated in 5.8ms.  In this case, the selection of sleep time becomes important to achieve gapless sample processing.

> In the end, it seems you really wanted a high priority task

Actually, what I really want is a simpler implementation that removes the need to poll the DMA memory in the application code.  I'd like to consider a more elegant and simpler solution.  The use of the ms_timer is a short-term fix to solve two issues:  high latency in coro switching (might be better with the new uasyncio?), and the risk of being blocked by multiple waiting slow coros.  There is likely a better way.  The I2S C-based module I wrote is going to be re-worked to implement the stream protocol, much like the existing UART module.  Then, ioctl can be used in the usasyncio scheduler to poll for new samples.  The benefits of this approach are described here:  [Writing streaming device drivers](https://github.com/peterhinch/micropython-async/blob/master/TUTORIAL.md#64-writing-streaming-device-drivers).  Then, the application just needs to make one simple call to acquire a block of samples.  For example, a call to `numread = await audio.readinto(buf)`.  The uasyncio scheduler handles the polling of the I2S peripheral and returns samples when they appear in DMA memory. 

> WHat does the high priority coro for the UART do? Is that the one for the pm sensor? That's a 30-something byte packet at 9600 baud every half second, no?

You're right - the coro is used for the particulate sensor, to read the UART at 9600 BAUD.  For this application, I'm uncertain if a high priority coro provides a meaningful performance gain.  The micropython module that I selected for the particulate sensor implements StreamReader to read the ESP32 UART.   I didn't do any performance tests - the PM module "just worked".

peterhinch, 2020-04-04:
@miketeachman It looks like pms5003.py does not use the StreamReader (`with_async=False`). It uses synchronous code to read the UART. Consequently UART processing will gain no benefit from **fast_io**.

kevinkk525, 2020-04-04:
@peterhinch that is not entirely true. pms5003 uses synchronous code if the device is in passive mode (you have to actively request data) but it uses the StreamReader in active mode (the device sends new data in its own interval).
I could have probably implemented it differently and always used StreamReader, I can't remember why I chose to make that difference. Probably to save asyncio Tasks and because cancelling a StreamReader was not possible.

tve, 2020-04-04:
> @tve great analysis of the code. It works as you indicated.

Thanks for the confirmation. What I learned from this example:
- The notion of task priorities is useful in use-cases where the driver doesn't provide any event/signal and only provides poll functionality (I'm using the generic meaning of event/signal/poll): in this example the run_mic needs to make progress even if other slow tasks happen to be runnable all at the same time
- A high-priority task may be a total pig and that's OK, by which I mean that the high priority task may block for an eternity (in this example on sdcard write) stalling out everything else, and that's OK in this example because the timing of everything else doesn't matter. (I my mind I tend to associate high-priority with short tasks.)

An interesting thought experiment might be to contemplate the following changes to the project:
- use fatfs or littlefs to write to the sdcard
- add a button with SW debounce to change the app behavior (start/stop or something else)

The only way to add SW debounce without changing everything else would be to do it with (soft) interrupts. That's because the high priority task blocks in sdcard write for a long time.

If one were to await in run_mic while the sdcard I/O is happening (I haven't looked into the details, I did spot a way to wait for a completion interrupt in esp-idf) then SW debounce using a high-priority coro might be feasible, might have to also add a couple of `await sleep(0)` "yield" statements into run_mic where it spends significant time in compute stuff.

If one were further to use a filesystem, then the calls to sdio would be buried in the filesystem code. That would make it impractical to specify what their priority should be. A high-priority task would "naturally" carry the priority through into the FS code, assuming that doesn't hand off the write to a separate task/coro internally. Of course at that point priority inversion, priority inheritance, etc. all start lurking for a chance to pounce... If one could specify the sdcard I/O or filesystem as high-priority stream then that would avoid the priority inversion and related problems.

So far my summary would be:
- task priorities are great because it's generally activities that have priorities and activities tend to map well to tasks, the downside are all the well-known issues with tasks and priorities
- I/O stream priorities are great because the priority applies to the I/O that "needs to keep moving" and it doesn't really matter whether an I/O access is by a high or low priority activity (which is good because the high priority activity's I/O may end up queued behind the low one), the downside is that one has to locate and "annotate" all the possible suspension points of a high priority activity and that is difficult.

peterhinch, 2020-04-05:
With a cooperative scheduler, priority inversion can be avoided by never yielding during a lock. This removes the need for priority inheritance. Locks carry many other hazards. Their use requires great care.

Blocking drivers such as sdcard play havoc in cooperative systems: sometimes the only solution is to use threading.

miketeachman, 2020-05-04:
@tve 
thanks for the suggestions on this project!  (I just noticed your followup this morning).  This project is on the backburner at the moment, but when it starts up again I'll re-visit your suggestions and see if I can apply them to the design.  Thanks again.  I really appreciate it :)