micropython/micropython #5858: extmod/uasyncio Please provide CPython's synchronisation primitives

peterhinch, 2020-04-02:
CPython provides Queue, Semaphore, BoundedSemaphore and Condition in addition to the provided Lock and Event classes.

In the case of Queue the issue was raised in https://github.com/micropython/micropython/issues/5828.

haldark, 2023-07-15:
I have installed nightly build for Wiznet 5100S-EVP-PICO.

dir(uasyncio) shows only the following. Still no 'Queue' class or 'queues' module. 

```
MicroPython v1.20.0 on 2023-04-26; W5100S-EVB-Pico with RP2040
Type "help()" for more information.
>>> import uasyncio
>>> for x in sorted(dir(uasyncio)): print(x)

CancelledError
Event
IOQueue
Lock
Loop
SingletonGenerator
StreamReader
StreamWriter
Task
TaskQueue
ThreadSafeFlag
TimeoutError
__class__
__dict__
__file__
__getattr__
__name__
__path__
__version__
_attrs
core
create_task
current_task
event
funcs
gather
get_event_loop
lock
new_event_loop
open_connection
run
run_until_complete
select
sleep
sleep_ms
start_server
stream
sys
ticks
ticks_add
ticks_diff
wait_for
wait_for_ms
>>> 
```

Do I have install another external module to get Queue?

Is it the issue only with Wiznet build? I would guess wiznet would not remove these classes as a ethernet chip running on SPI port would not pose any risk to these classes.

Or, main build too does not have Queue any more?

peterhinch, 2023-07-16:
There is no official `Queue` module. See [this doc](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md) which documents unofficial solutions for the missing primitives.