micropython/micropython #5860: Arduino MKR Zero support?

simonvanderveldt, 2020-04-03:
First of all thanks for this awesome project!

Wanted to ask what would be necessary to add support for the Arduino SAMD21 based boards (specifically the MKR Zero)? Is there maybe already someone working on this?

The only relevant issues/PRs I could find are #4863 and #4994 but whilst they add SAMD support they don't contain anything for the Arduino boards.

Whilst I have no experience with this I don't mind creating a PR to add it myself, but I could use some help/input on what needs to be done to add support for a new board. I couldn't really find any documentation for it.

dpgeorge, 2020-04-04:
> Whilst I have no experience with this I could work on creating a PR to add it myself, but I could use some help/input on what needs to be done to add support for a new board.

The best place to start would be to take an existing board from `ports/samd/boards/` that matches the one you want and adapt it.

robert-hh, 2022-12-15:
Adapting for a new board is pretty easy. The most laborious part is changing pins.csv, but any firmware for a SAM21 board should run a the device. Only the board name and the pin names may be wrong, but you can use the port numbers to designate a pin, like Pin("PA04").

Carglglz, 2023-01-21:
@simonvanderveldt I had an old `MKRZERO` lying around and I've managed to implement this board. 
First you need to flash the UF2 bootloader (follow [this Adafruit guide](https://learn.adafruit.com/installing-circuitpython-on-samd21-boards/installing-the-uf2-bootloader) and then I used `ADAFRUIT_ITSYBITSY_M0_EXPRESS` as a template.
I changed the board name and `pins.csv`  to match the [pinout](https://docs.arduino.cc/static/eb25af1929df3a7a192dbc48d6ce6cbc/ABX00012-full-pinout.pdf) and [schematics](https://docs.arduino.cc/static/d9d41fff897606ce51c20412cd26ffab/schematics.pdf), compiled and uploaded `firmware.uf2` to `MRZEROBOOT` drive and that's it.
```
MicroPython v1.19.1-831-g4f3780a15-dirty on 2023-01-21; MKRZERO with SAMD21G18A
Type "help()" for more information.
```
I may make a PR but in the meantime check my branch [mkrzero](https://github.com/Carglglz/micropython/tree/mkrzero/ports/samd/boards/ARDUINO_MKRZERO)

robert-hh, 2023-01-22:
@Carglglz That's very good. The Arduino MKR Zero looks like a good board, with the SD card, the crypto chip and the available JTAG interface. Please make a PR, when you think you're ready.

Since the board has a 32kHz crystal, you could try to set in mpconfigboard.h

`#define MICROPY_HW_XOSC32K  (1)`
instead of:
`#define MICROPY_HW_DFLL_USB_SYNC    (1)`

That results in: 
- availability of machine.RTC()
- better clock precision when the board is not connected to USB
- lower jitter of machine PWM
- a slightly shorter start-up time.

When using `MICROPY_HW_DFLL_USB_SYNC`, the clock reference is the 1 kHz USB sync pulse, when connected, and the internal 32kHz RC oscillator, when USB is not connected. The latter has a quite high temperature drift. and using a 1 kHz sync signal for the 48MHz and 120Mhz oscillators causes more jitter than using the 32kHz signal.

robert-hh, 2023-01-22:
Just another note: Since the firmware supports touch1200, you may use the little script below to enable the bootloader:
```
import serial
import time
import sys
com=serial.Serial(sys.argv[1], 1200, dsrdtr=True)
com.dtr=False
com.rts=False
time.sleep(0.2)
com.close()
time.sleep(4)
```
It requires the `pyserial` module on the PC

Carglglz, 2023-01-22:
@robert-hh thanks, really nice work with the `samd` port 👏🏼 
> The Arduino MKR Zero looks like a good board, with the SD card, the crypto chip and the available JTAG interface.

Yes, so far I've only done a quick test with the led and `UART` and so far so good, SD and crypto chip will go next...

> Since the board has a 32kHz crystal, you could try to set in mpconfigboard.h

I wasn't aware of this, I did tried importing RTC which didn't work so thanks for the explanation 👍🏼.

Also any hint on checking what is the maximum firmware size for the `SAMD21G18A`?, Arduino docs says 
> 256KB Flash

but somehow the uf2 size is 
```
Converted to uf2, output size: 379392, start address: 0x2000
Wrote 379392 bytes to build-ARDUINO_MKRZERO/firmware.uf2
```
I'm missing something obvious I guess?

> Just another note: Since the firmware supports touch1200, you may use the little script below to enable the bootloader:

Thanks for this too, it would be nice to automate the flashing process tbh ⚙️

robert-hh, 2023-01-22:
The flash size id 256k. Of that, 192 k are reserved by the code and almost fully used, 64k for the file  The .uf2 files are about twice as large as the binary image itself. That's why you get the confusing number.
You can force the machine.RTC class to be included by setting in mpconfigboard.h:
`#define MICROPY_PY_MACHINE_RTC          (1)`

But without a crystal the RTC clock would be controlled the internal low power 32kHz RC oscillator. And even it if is quite good for an RC oscillator, it is not good enough for a clock. But if you need the RTC class for consistency, you can enable it.

simonvanderveldt, 2023-01-23:
@Carglglz Nice! I don't use the MKR Zero anymore, but good to see that this might still happen! :)

robert-hh, 2023-01-24:
At the end of the build process the linker tells the size of the binary. Like with these lines:
```
   text	   data	    bss	    dec	    hex	filename
 184596	    148	   1788	 186532	  2d8a4	build-ARDUINO_MKR_ZERO/firmware.elf
```
The 186532 plus 0x2000 is the total size in flash which is required. So it's 190k at the moment. 0x2000 is the start address in flash.
About SD-Card: The SAMD port does not support FAT at the moment. That's why you get the `OSError 19` which tells, that the file system is not known. You could try to enable FAT by:
- enabling the lines below to `mpconfigboard.mk`
```
MICROPY_VFS_FAT ?= 1
SRC_C += \
	fatfs_port.c
```
- adding the FAT settings to `mpconfigboard.h`:
```
// fatfs configuration used in ffconf.h
#define MICROPY_FATFS_ENABLE_LFN            (1)
#define MICROPY_FATFS_RPATH                 (2)
#define MICROPY_FATFS_MAX_SS                (4096)
#define MICROPY_FATFS_LFN_CODE_PAGE         437 /* 1=SFN/ANSI 437=LFN/U.S.(OEM) */
```
This will increase however the space for the code increases to 204k. So you have reduce the size of the on-board file system by changing boards/samd21x18a.ld e.g. to the lines below:
```
_oflash_fs = ORIGIN(FLASH) + 192K + 16K - 8K;
_sflash_fs = LENGTH(FLASH) - 192K -16K + 8K - 1;
```
Which decrease the file system to 48k and increase the space for the code by 16k. You can save that to a specific .ld file and reference that in the board's `mpconfigboard.mk`. At the moment, there is not warning if the code overlaps the file system. The decrease of the file system may be acceptable if you have only a small script there which mounts the SD card. Note that FAT also allocates a 4k buffer from RAM.

b.t.w.: Did you try enabling the crystal?

robert-hh, 2023-01-24:
About the crypto chip: On the picture of the board at arduino.cc I see a non-populated chip area between the MCU and the SD card connector. Most likely that the place where the crypto chip ought to be. The connections look right.

Carglglz, 2023-01-24:
@robert-hh  
> About SD-Card: The SAMD port does not support FAT at the moment. That's why you get the OSError 19 which tells, that the file system is not known

Well that makes sense 😅 , I will try that next...

> b.t.w.: Did you try enabling the crystal?

Yes,  it did work and `RTC` too 👍🏼 , I will update whether I manage to mount the SD or not.

> bout the crypto chip: On the picture of the board at arduino.cc I see a non-populated chip area between the MCU and the SD card connector. Most likely that the place where the crypto chip ought to be. The connections look right.

Yes I can see the footprint now you've mentioned it. I guess it could be soldered but I have no `ATECC508A` chip at the moment to try...

Carglglz, 2023-01-30:
@robert-hh FAT enabled and SD mounted successfully 🎉 . I will do the PR soon.