micropython/micropython #5864: Bug: select, poll and ipoll timeouts subject to overflow

george-hawkins, 2020-04-05:
The timeouts for `select`, `poll` and `ipoll` (in `uselect`) don't take into account the potential for overflow. They should use the same logic that's already used by `utime.ticks_diff`.

If you look at [`moduselect.c:248`](https://github.com/micropython/micropython/blob/69661f3/extmod/moduselect.c#L248) you see:

    mp_uint_t start_tick = mp_hal_ticks_ms();
    ...
    for (;;) {
        ...
        if (n_ready > 0 || (timeout != -1 && mp_hal_ticks_ms() - start_tick >= timeout)) {

But `mp_hal_ticks_ms` may have overflowed between using it to set `start_tick` and using it in the if-condition.

This logic should instead use the same kind of ticks-diff logic that's already seen in `utime`. Here we see [`time_ticks_diff`](https://github.com/micropython/micropython/blob/69661f3/extmod/utime_mphal.c#L82) defined as:

    mp_int_t diff = ((end - start + MICROPY_PY_UTIME_TICKS_PERIOD / 2) & (MICROPY_PY_UTIME_TICKS_PERIOD - 1))
        - MICROPY_PY_UTIME_TICKS_PERIOD / 2;

I.e. a somewhat more complicated rule is needed to account for the wraparound that may occur.

In Python, one would rewrite the timeout check in `moduselect.c` as:

```Python
import time

start_tick = time.ticks_ms()

while True:
        if n_ready > 0 or (timeout != -1 and time.ticks_diff(time.ticks_ms(), start_tick) >= timeout):
```

george-hawkins, 2020-04-05:
Doing a quick grep for "timeout" and " - " I see this style of timeout check in lots of places:

```
extmod/machine_pulse.c:       if ((mp_uint_t)(mp_hal_ticks_us() - start) >= timeout_us) {
extmod/machine_pulse.c:       if ((mp_uint_t)(mp_hal_ticks_us() - start) >= timeout_us) {
extmod/nimble/nimble/npl_os.c: while (sem->count == 0 && mp_hal_ticks_ms() - t0 < timeout) {
extmod/modlwip.c:             if (socket->timeout != -1 && mp_hal_ticks_ms() - start > socket->timeout) {
extmod/modlwip.c:             if (socket->timeout != -1 && mp_hal_ticks_ms() - start > socket->timeout) {
ports/esp8266/uart.c:         if (system_get_time() - start >= timeout_us) {
ports/teensy/core/usb_serial.h: } while(millis() - startMillis < _timeout);
ports/stm32/usbd_cdc_interface.c: if (HAL_GetTick() - start >= timeout) {
ports/stm32/usbd_cdc_interface.c: if (HAL_GetTick() - start >= timeout) {
ports/stm32/pyb_i2c.c:        if (HAL_GetTick() - start >= timeout) {
ports/stm32/rtc.c:            if ((HAL_GetTick() - tickstart ) > timeout) {
ports/stm32/spi.c:            if (HAL_GetTick() - t_start >= timeout) {
ports/stm32/can.c:            if (HAL_GetTick() - start >= timeout_ms) {
ports/stm32/fdcan.c:          if (HAL_GetTick() - start >= timeout_ms) {
ports/stm32/sdcard.c:         if (HAL_GetTick() - start >= timeout) {
ports/stm32/sdcard.c:         if (HAL_GetTick() - start >= timeout) {
ports/stm32/i2c.c:            if (HAL_GetTick() - t0 >= i2c_timeout_ms[i2c_id]) {
ports/stm32/i2c.c:            if (HAL_GetTick() - t0 >= i2c_timeout_ms[i2c_id]) {
ports/stm32/i2c.c:            if (HAL_GetTick() - t0 >= i2c_timeout_ms[i2c_id]) {
ports/stm32/i2c.c:            if (HAL_GetTick() - t0 >= i2c_timeout_ms[i2c_id]) {
ports/stm32/usbd_hid_interface.c: if (mp_hal_ticks_ms() - t0 >= timeout_ms || query_irq() == IRQ_STATE_DISABLED) {
ports/stm32/uart.c:           if (HAL_GetTick() - start >= timeout) {
ports/stm32/uart.c:           if (HAL_GetTick() - start >= timeout) {
ports/stm32/uart.c:           if (timeout == 0 || HAL_GetTick() - start >= timeout) {
ports/stm32/rfcore.c:         if (mp_hal_ticks_ms() - t0 > timeout_ms) {
ports/stm32/rfcore.c:         if (mp_hal_ticks_ms() - t0 > timeout_ms) {
ports/stm32/machine_adc.c:    if (mp_hal_ticks_ms() - t0 > timeout_ms) {
ports/stm32/adc.c:            if (((HAL_GetTick() - tickstart ) > timeout)) {
```

dpgeorge, 2020-04-05:
> Doing a quick grep for "timeout" and " - ", I see this style of timeout check in lots of places

That's right, it's used in lots of place.  It does take account of the overflow because it's doing unsigned arithmetic (or at least it should be): `t1 - t0` is unsigned and so always represents the correct (non negative) time delta.

george-hawkins, 2020-04-05:
Well, that's embarrassing - clearly, I've forgotten how to think about unsigned values. I'd been using `ticks_diff` in my Python code and looking at this C code I thought there was an issue. I had to write a piece of C to convince myself that this worked out as desired:

```C
#include <stdio.h>

int main(void) {
  unsigned start = -3;
  unsigned now = start;

  for (int i = 0; i < 5; i++) {
    printf("start=%u, now=%u, (now - start)=%u\n", start, now, (now - start));
    now++;
  }
  return 0;
}
```

As expected the output (on a machine where `unsigned` is 32 bits) is:

```
start=4294967293, now=4294967293, (now - start)=0
start=4294967293, now=4294967294, (now - start)=1
start=4294967293, now=4294967295, (now - start)=2
start=4294967293, now=0, (now - start)=3
start=4294967293, now=1, (now - start)=4
```
Sorry for wasting your time. And thanks for MicroPython - it's super :slightly_smiling_face:

closed 2020-04-05 by george-hawkins

davehylands, 2020-04-06:
It also works with signed integers. What's significant about the behavior is that it wraps. So for unsigned 0xffffffff + 1 gives you 0, and for signed 0x7fffffff + 1 = 0x80000000, i.e. largest positive + 1 = largest negative. In Python, this wrap doesn't occur, so you need to do an explicit mask to force it. Since micropython has multi-precision integers, it just changes from a small int to an MPZ int.