micropython/micropython #5865: bug in evaluating const

dugres, 2020-04-05:
>>> a = const(2*8)
>>> b = const(2**8)
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: constant must be an integer

jimmo, 2020-04-05:
`const()` can only work on either literal integers, or expressions that can be constant-folded to literal integers.

Constant folding supports the following operators: |, &, ^, <<, >>, +, -, %, //  (and unary +, -, ~). This isn't so much a bug (there's an explicit comment `// Can't fold @ or / or **` in the code that handles this), as maybe a limitation. I don't know the exact reason, but I would imagine that the additional code to ensure that it's a case that resulted in a small int was too costly (and given that shifting is supported probably wasn't deemed necessary).

So in your case you could use `const(1 << 8)` to achive the same thing.

dpgeorge, 2020-04-06:
That is correct, power is not supported in const expressions.  It could be but it would cost extra code, and it's likely that `<<` is enough.

dugres, 2020-04-06:
If it's the intended behavior, it's ok, The error message is miss leading though.

dpgeorge, 2020-04-09:
> The error message is miss leading though.

It's currently "constant must be an integer", which might be slightly clearer if changed to "bad const expression".

dugres, 2020-04-09:
I think that "unsupported const expression" might be even clearer.

dpgeorge, 2020-04-09:
See #5891 for a PR that implements folding of power.

dpgeorge, 2020-04-09:
> I think that "unsupported const expression" might be even clearer.

Yes it would, but also need to think about code size, and longer messages increase code size.  But here maybe it's worth a few extra bytes.

dpgeorge, 2020-05-11:
Power operator is now supported in const expressions (see commit above).

closed 2020-05-11 by dpgeorge

louis-riviere-xyz, 2020-05-11:
Very nice.
Improving consistency is a good thing.
👍