micropython/micropython #5866: Error when building nRF PCA10059 (nRF52840 dongle)

scttnlsn, 2020-04-05:
From within the `ports/nrf` directory I'm running:

```
make BOARD=pca10059
```

And getting the following error:

```
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build-pca10059/genhdr
Create build-pca10059/genhdr/pins.h
GEN build-pca10059/genhdr/mpversion.h
GEN build-pca10059/genhdr/moduledefs.h
make: *** No rule to make target 'lib/tinyusb/src/portable/nordic/nrf5x/hal_nrf5x.c', needed by 'build-pca10059/genhdr/qstr.i.last'.  Stop.
```

Building other targets (i.e. `pca10028`) works fine.  Any ideas?

hoihu, 2020-04-05:
I can confirm this. I think it is related to the tinyusb‘s use of the nrfx lib (which has been updated upstream and is a different version than the on in the nordic port...).

That is why only the pca10059 is affected, it is the only one that uses the usb module of the nrf52840. 

So two modules using a different version of the hal is likely causing troubles... not sure how to solve it. Perhaps either go back in time using git bisect or try updating the nrfx lib in the nordic port and crawl through the compiler errors...

jimmo, 2020-04-05:
Please see #5791 for a fix for this.

pacmac, 2020-04-19:
is there a permanent fix for this without needing to use a branch ?

dpgeorge, 2020-04-20:
> is there a permanent fix for this without needing to use a branch ?

Best thing to do is use a stable release version.