micropython/micropython #5867: ESP32: BLE Compatibility Issue Advisory from Espressif

jacklinquan, 2020-04-05:
This morning my company received this.
I'm not sure if micropython esp32 port is influenced, please be aware of it.

兼容性问题公告
Compatibility Issue Advisory
Title 标题
苹果 iOS 13.4 系统和部分 ESP32 产品 BLE 连接的兼容性问题
BLE connection breaks immediately between devices in iOS 13.4 and some ESP32-based
products due to compatibility issue
1. Issue Summary 问题⼩小结
近期，乐鑫接到了了部分⽤用户的反馈，苹果⼿手机升级到 iOS 系统 13.4 后与部分 ESP32 设备有兼容性问题：
1. 主要现象：升级系统后，ESP32 与苹果⼿手机⽆无法连接成功；
2. 当前终端⽤用户上报发现是使⽤用 DIS (Device Information Service) 的设备有兼容性问题。
Recently, some end users feedback that their iPhones using iOS 13.4 are not compatible with partial ESP32-
embedded devices:
1. Problem: Cannot connect ESP32 based devices with iPhones updated to iOS version 13.4.
2. Devices with compatibility issues reported are those using DIS (Device Information Service).
2. Reproduction/Verification 问题复现/验证
终端客户复现⽅方式：将苹果 iOS 系统升级 13.4 后， 苹果⼿手机蓝⽛牙连接上 ESP32 后，很快（ 约 10 秒后）
会⾃自动断开。
乐鑫团队复现⽅方式： 参考 BLE DIS (Device Information Service) 协议, 基于 ESP-IDF 的Bluedroid Host 的
APIs, 使⽤用 attribute table ⽅方式新建⼀一个 DIS Service. 然后使⽤用苹果 iOS 13.4 ⼿手机和 ESP32 设备 进⾏行行连接,
iOS 设备会很快断开连接（约 10 秒后）。
Symptoms reported: BLE connection between the ESP32-embedded device and iPhone updated to iOS
13.4 is automatically broken shortly after (about 10 seconds) establishment.
Reproduction by ESPRESSIF: Referring to BLE DIS (Device Information Service) protocol, we used an
attribute table to create a DIS Service based on Bluedroid Host APIs in ESP-IDF and connected the ESP32
based device with iPhone in iOS 13.4. The connection was broken in about 10 seconds.
3. Impact on Customer 客户端影响分析
Affected features:
1. Customers who develop applications based on Bluedroid Host and implement some Bluedroid-based
services may be affected if end users use iPhones in iOS 13.4. For example if customer implements DIS
service using Bluedroid GATT API, then due to modified behaviour of iOS in version 13.4, the discovery of
the device will fail.
2. Customers using ESPRESSIF BluFi provisioning service won't be affected as BluFi does not include DIS
service internally by default. However if you have added DIS service support in BluFi on your own, the
devices will see the impact.
3. Customers who develop applications using other ESP32 BLE features using Bluedroid GATT API may
also be affected.
4. Customers who use NimBLE Host won’t be affected by the issue. Hence customers using
ESPRESSIF Unified Provisioning over BLE won’t be affected.
4. Root Cause 根因
由于苹果 iOS 13.4 系统更更改了了 BLE Service Discovery 流程, 在发现 DIS 等服务的时候, 除了了常规的发现过
程, 还会⽤用 Read By Type Request 来读取特定 UUID 对应 Characteristic 的值，乐鑫 ESP-IDF 需配合其进⾏行行
相应的软件处理理机制优化。
Apple has changed BLE Service Discovery procedures in iOS 13.4. Apart from common procedures to
discover DIS services, iOS 13.4 uses Read-By-Type Request to read characteristic values corresponding to
certain UUID. In response, our team needs to optimize related software mechanisms on ESP-IDF.
5. Solution & Support 解决⽅方法 & ⽀支持
1. 强烈烈建议⽤用户进⾏行行⾃自查，如使⽤用 ESP32 蓝⽛牙配⽹网服务且 BLE 使⽤用包含 DIS Service（基于 Bluedroid
Host），需进⾏行行 ESP-IDF 软件升级与⽤用户端的固件升级；乐鑫团队⽬目前已经针对 IDF 维护版本，提供
patch ，通⽤用版本的 IDF 合⼊入情况与对应版本请参考 6.附录。
2. 建议使⽤用 ESP32 Bluedroid BLE 功能的客户进⾏行行⾃自查，按照如上 2. 问题复现/验证 进⾏行行排查，确认 iOS
13.4 是否有兼容性问题；如有，请进⾏行行 ESP-IDF 软件升级与⽤用户端的固件升级；
3. 如需软件升级⽀支持、问询，请联系乐鑫⽀支持团队 或 sales@espressif.com。
1. It is highly recommended that customers who use ESP32 BLE provisioning and Bluedroid-based DIS
Service upgrade ESP-IDF and firmware. Our team has provided a patch for IDF release versions and
merge into released versions are under progress. Merge details in released version as Section 6
Appendix.
2. It is recommended that customers using ESP32 Bluedroid BLE check for compatibility issues according
to Section 2 Reproduction/Verification. If compatibility issues are found, please upgrade ESP-IDF and
firmware;
3. For software upgrade support and queries, please contact ESPRESSIF support team or
sales@espressif.com.
6. Appendix 附录
如下是截⽌止到 2020 年年 4 ⽉月 3 ⽇日，ESP-IDF 通⽤用版本 IDF 合⼊入情况：
Below is the Merge details of Released Branches in ESP-IDF (up to Apr.3,2020):
Branch Commit ID 版本 Commit ID Merge Status (合⼊入状
态）
Master dcc2963d431ef7eccc4c0abd0a95781698bb4e75 Synced to Github
release/v4.1 5e09d473c70367ce3899706db48ce3313162c0c4 Synced to Github
release/v4.0 f69f96edab61dddc13ca9d36ea00e3b0d91f38dc Under Merge process
release/v3.3 a71e85fb63b9549fd04b5e99fe1d345b2e628d9d Under Merge process
release/v3.2 caacfd87fae40023c370451d69b2138f36e68974 Synced to Github

jimmo, 2020-04-05:
> (4). Customers who use NimBLE Host won’t be affected by the issue. Hence customers using
ESPRESSIF Unified Provisioning over BLE won’t be affected.

MicroPython only uses the NimBLE host, not BlueDroid.

jacklinquan, 2020-04-06:
Thanks jimmo!
I'll keep this issue open for some time and then close it.

closed 2020-05-27 by jacklinquan

curlyz, 2023-05-13:
Does we have any luck on connecting between iOS devices and ESP32 BLE mode ?
It all seems like the core reason is that Apple does't support SPP. But then I am wondering how the module HM-10 are able to do so.
Here is list of links that I have come across, do we have an micropython examples that allow iPhone to connect to ESP32 ? Much thanks

[https://github.com/micropython/micropython/issues/5867](https://github.com/micropython/micropython/issues/5867)

[https://esp32.com/viewtopic.php?t=27483](https://esp32.com/viewtopic.php?t=27483)

[https://esp32.com/viewtopic.php?f=13&t=30702](https://esp32.com/viewtopic.php?f=13&t=30702)

[https://developer.apple.com/forums/thread/698726](https://developer.apple.com/forums/thread/698726)

[https://developer.apple.com/forums/thread/692718](https://developer.apple.com/forums/thread/692718)

[https://stackoverflow.com/questions/75577202/ios-ble-and-esp32-ble-communication-is-not-working](https://stackoverflow.com/questions/75577202/ios-ble-and-esp32-ble-communication-is-not-working)

[https://support.apple.com/en-vn/HT201205](https://support.apple.com/en-vn/HT201205) 

[https://www.esp32.com/viewtopic.php?t=10136](https://www.esp32.com/viewtopic.php?t=10136)

[https://support.apple.com/en-us/HT204387](https://support.apple.com/en-us/HT204387)

[https://github.com/espressif/esp-nimble/issues/33](https://github.com/espressif/esp-nimble/issues/33)

[https://github.com/rusel1989/react-native-bluetooth-serial/issues/130](https://github.com/rusel1989/react-native-bluetooth-serial/issues/130)