micropython/micropython #5870: extmod/uasyncio: add serve_forever()?

belyalov, 2020-04-06:
Official example from python documentation:
```python
async def main():
    # Get a reference to the event loop as we plan to use
    # low-level APIs.
    loop = asyncio.get_running_loop()

    server = await loop.create_server(
        lambda: EchoServerProtocol(),
        '127.0.0.1', 8888)

    async with server:
        await server.serve_forever()

asyncio.run(main())
```

Micropython's `Server` class already has support for `async with`, but lacks convenient method `serve_forever`.

Do you think it is make sense to add it?

dpgeorge, 2020-04-07:
> Do you think it is make sense to add it?

What is the use-case you are thinking of?  The existing `uasyncio.start_server()` does effectively serve forever until the `Server` instance is closed.

For reference the CPython issue discussing `serve_forever` and related functionality is here: https://bugs.python.org/issue32662

belyalov, 2020-04-07:
Mm, could you please provide an example how to use it then?
It is a bit unclear:
```python
await with uasyncio.start_server():
    # ?? some "blocking call"?
```

dpgeorge, 2020-04-07:
> could you please provide an example how to use it then?

This might work:
```python
server = await uasyncio.start_server()
await server.wait_closed()
```
But I'm not sure if that would also work under CPython...