micropython/micropython #5871: TimeoutError is missing from builtins

matejcik, 2020-04-06:
it is possible to enable `MICROPY_PY_BUILTINS_TIMEOUTERROR` in mpconfigport, and then objexcept.c [will declare] a `TimeoutError`.

The corresponding piece [seems to be missing] from modbuiltins.c

[will declare]: https://github.com/micropython/micropython/blob/master/py/objexcept.c#L257-L259
[seems to be missing]: https://github.com/micropython/micropython/blob/master/py/modbuiltins.c#L736-L770

dpgeorge, 2020-04-07:
> The corresponding piece [seems to be missing](https://github.com/micropython/micropython/blob/master/py/modbuiltins.c#L736-L770) from modbuiltins.c

Yes that's true, it's not available as a built-in like other exceptions are.  No code actually uses this exception type.

Adding exceptions to the firmware takes code size.  In this case it's recommended to use `OSError(ETIMEDOUT)` to indicate a timeout error.  (`TimeoutError` is  a subclass of `OSError`.)

What is your use-case for `TimeoutError`?

matejcik, 2020-04-07:
I mean, the big Python has it...? :)

No, my point is, declaring `mp_type_TimeoutError` but then not publishing it in builtins (what we have now) doesn't make much sense?
Either `MICROPY_PY_BUILTINS_TIMEOUTERROR` shouldn't exist, or `TimeoutError` should be visible in builtins.

dpgeorge, 2020-04-07:
> Either `MICROPY_PY_BUILTINS_TIMEOUTERROR` shouldn't exist, or `TimeoutError` should be visible in builtins when it is enabled.

Yes I agree with this.  So maybe it should be removed?

It was added back in June 2015 in 077812b2abe3f5e5325194f9694dad7eb38186dd for the cc3200 port.  In f522849a4d5a978ac3d322d71a755f75d07e8ce6 the cc3200 port enabled use of it in the socket module aliased to `socket.timeout`.  That's why it was never added to the builtins.  Then it was replaced by `OSError(ETIMEDOUT)` in Feb 2017 in 047af9b10bfc6b0ec412f8450c6bec10ab95254b.

The esp32 port enables this exception, since the very beginning of that port, but it could never be accessed because it's not in builtins.

So I'd say it's safe to just remove it completely.

matejcik, 2020-04-07:
That is fine by me.
(for my usecase it is nice to have a readily-available builtin, but I can just as easily declare it in mpconfigport, or use a custom exception class instead)

dpgeorge, 2020-04-07:
> for my usecase it is nice to have a readily-available builtin, but I can just as easily declare it in mpconfigport, or use a custom exception class instead

Note that ports can add anything to the builtins via `MICROPY_PORT_BUILTINS`.  And they can also define their own exceptions using the `MP_DEFINE_EXCEPTION()` macro.  Or it could be done in pure Python quite easily (and inserted into the builtins from Python).

I think it's a good idea to remove it from the core: 1) to not encourage its use; 2) because there are a lot of other OSError subclasses (eg `BlockingIOError`) which are not defined at all, and having `TimeoutError` is a bit inconsistent.

dpgeorge, 2020-04-07:
See #5876 for a PR to remove `TimeoutError`.

dpgeorge, 2020-04-09:
`TimeoutError` and `MICROPY_PY_BUILTINS_TIMEOUTERROR` were removed completely in e292296d52800939e70513e33ec456518330c4ad

dpgeorge, 2020-04-09:
@matejcik thanks for raising this issue, it's always nice to clean up / remove things from the code.

closed 2020-04-09 by dpgeorge