micropython/micropython #5872: Discrepancy between test running and cleaning/printing failures

stinos, 2020-04-06:
The run-tests script puts the test output in the current working directory but the --print-failure and --clean-failure arguments look for test output in the tests/ directory. As a result the two only work together if the working directory for running the tests is effectively the tests/ directory. I found this out when running tests by calling run-tests by full path from wherever the cwd is: it would never print failures.

It would be better if the script would stick to one thing: either read test output from the cwd as well, or else produce it in the tests/ directory (which would be my favorite: not a fan of relying on cwd except for commandline usage). The latter could be extended by allowing to specify the output directory as an argument. Ideas of which way to go?

dlech, 2020-04-06:
I would be in favor of having a default output directory that is excluded from git, similar to the `build` directory for compiler output, e.g. `tests/results/` or `tests/output/`.

dpgeorge, 2020-04-07:
> The latter could be extended by allowing to specify the output directory as an argument

+1

> I would be in favor of having a default output directory that is excluded from git, similar to the `build` directory for compiler output, e.g. `tests/results/` or `tests/output/`.

+1

stinos, 2020-04-07:
Note there's a similar problem with the default values of MICROPY_MICROPYTHON and MPYCROSS in run-tests: they are relative to the working directory so if you don't invoke run-tests from within it's own directory you have to specify them because the defaults are not correct anymore.

stinos, 2020-04-07:
I started on this but turns out it is actually quite involved to the point I'm not sure if it's even worth it: there's really a lot of things which expect they're ran relative to the tests directory. Get a glimpse here: https://github.com/stinos/micropython/tree/run-test-abspath (note: still doesn't work). Maybe there's an easier way? But even if it would be fixed, all other test runners suffer the same problem. Perhaps just making it required for the test runners to run from the tests directory and raising an error if not would be better..

commit 405893af (2020-08-27): tests/run-tests: Use absolute paths where possible.

Replace some usages of paths relative to the current working directory
with absolute paths relative to the tests directory.

Fixes and resulting changes:
- default values of MICROPYTHON and MPYCROSS are absolute paths and
  always correct
- likewise, the correct full paths for tools and extmod directories
  are appended to sys.path
- printing/cleaning failures works properly since it expects the .exp
  and .out files in the tests directory which is also where they
  are written to now, plus no more need for changing directories

This fixes # [... cut]
closed 2020-08-27 by dpgeorge